Ultimaker/Cura #9988: LAST CURA VERSIONS (4.9.0 / 4.9.1 CRASHES IN WINDOWS 10

MONarqs, 2021-06-16:
### Application Version

4.9.1+ 4.9.0

### Platform

windows 10.0.19041

### Printer

Ultimaker S5

### Reproduction steps

1. Installed last version of CURA 4.91
2.  Fatal error shows up after trying to import any model to CURA
3. It closes (and everytime I try to open different files used before in 4.8.0 it gives me the same FATAL ERROR...(it opens 10 crash report windows with the same info but it doesn't let me go on with anything inside CURA, I try to send the report but nothing happens and if Iclose the window report it closes CURA).

4. Therefore I install the previous version of CURA 4.9.0
5.  Fatal error shows up after trying to import any model to CURA
6. It closes (and everytime I try to open different files used before in 4.8.0 it gives me the same FATAL ERROR...(it opens 10 crash report windows with the same info but it doesn't let me go on with anything inside CURA, I try to send the report but nothing happens and if Iclose the window report it closes CURA).

7. I reinstall CURA 4.8.0 and it works fine!

...SO THERE'S SOMETHING WRONG WITH LAST 2 VERSIONS. PLEASE SEE FILES ATTACHED...

[image]

[cura.log](https://github.com/Ultimaker/Cura/files/6660362/cura.log)

### Actual results

After installed last versions (4.9.0 + 4.9.1) both crash suddenly after importing a file and CURA closes!

### Expected results

I hope you can fix it because I had to go back to version  4.8.0

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

info crash:

Traceback (most recent call last):
  File "C:\Users\monar\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 120, in _triggerJobNameUpdate
    self._updateJobName()
  File "C:\Users\monar\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 129, in _updateJobName
    base_name = self._print_information._stripAccents(self._print_information._base_name)
AttributeError: 'PrintInformation' object has no attribute '_stripAccents'

second info window crash:

Thread 0x0000361c (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003ca4 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003ca8 (most recent call first):
[... 154 lines cut ...]
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00006698 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x00000d44 (most recent call first):
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 167 in exceptHook
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\UI\PrintInformation.py", line 253 in _calculateInformation
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\UI\PrintInformation.py", line 169 in _onPrintDurationMessage
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\UI\PrintInformation.py", line 428 in setToZeroPrintInformation
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\UI\PrintInformation.py", line 441 in _onSceneChanged
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "X:\4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "X:\4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

nallath, 2021-06-16:
This is an issue in the CustomJobPrefix plugin. Please update that.

closed 2021-06-16 by nallath

MONarqs, 2021-06-16:
I don't know how to do it. Could you please help me?

Thanks in advance,

Yours,

Pablo Perlado

nallath, 2021-06-16:
Go to the marketplace in Cura, find the plugins that can be updated in the "installed" tab, click the update button

MONarqs, 2021-06-16:
Thanks so much…just iupdated and everything works FANTASTIC as usual.

 

 

Pablo

 

Pablo Perlado, architect

 

                                                                       MON ARQUITECTOS

 

 

MON ARQUITECTOS SL

 

Architecture + Urban Planning

 

Calle Alonso Cano 99, 4ºB        

 

28003-MADRID (SPAIN)   

  

 <http://www.monarquitectos.com/> www.monarquitectos.com 

 

studio..  +34 649741252       

mobile.  +34 649741252  

 ***@***.***> ***@***.***

 

………….................................................................please consider the enviroment before printing this e-mail

Enviado el: miércoles, 16 de junio de 2021 13:04
Para: Ultimaker/Cura ***@***.***>
CC: MONarqs ***@***.***>; Author ***@***.***>