Ultimaker/Cura #9994: No Nozzle Option or Profiles if created a new definition

Yijare, 2021-06-17:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 6

### Reproduction steps

I created a new definition for the Creality Ender 6, as Cura does not provide this one - took creality_ender5.def.json and rewrote the contents there in

### Actual results

No Nozzle options avaible - 'empty' is displayed instead

[image]

No Profiles for Quality are available - or creatable

[image]

### Expected results

Nozzle Options and Profiles should apear like on the Ender 5, where I took the base def.json from to create the Ender 6 def.json

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I know I could just create myself an Ender 5 and rewrite the Printer setting in Cura itself, but found it more convinient to make a definition instead
[creality_ender6.def.zip](https://github.com/Ultimaker/Cura/files/6673465/creality_ender6.def.zip)

fvrmr, 2021-06-18:
Hi @Yijare thank you for your report. 
You need to add the nozzle profiles in `variants` folder.
You can copy the profiles from the Ender 5 and change the name of the files to Ender 6: `creality_ender6_0.4.inst.cfg` and edit the definition in the files to `definition = creality_ender6`

I would consider this as a duplicate of https://github.com/Ultimaker/Cura/issues/8882
I will close this issue as solved, since the solution should work. But feel free to reach out if you have more questions.

closed 2021-06-18 by fvrmr

Yijare, 2021-06-18:
oh, thank you. oversight on my part then.