Ultimaker/Cura #10000: Cura is Cool

fieldOfView, 2021-06-18:
### Is your feature request related to a problem?

no

### Describe the solution you'd like

Cura

### Describe alternatives you've considered

none, really

### Affected users and/or printers

any

### Additional information & file uploads

Feel free to close this as a duplcuate of #4000

fieldOfView, 2021-06-18:
10.000 issues in the queue
10.000 issues.
Take one down, pass it around
10.001 issues on the wall!

rburema, 2021-06-22:
Awesome!

Fun fact: 'Issues' and 'pulls' share the number-list, so we could make the next round number a PR as well :-p

fieldOfView, 2021-06-22:
'Issues' and 'pulls' AND 'discussions'!

I wonder if anything special happens at issue/pull/discussion 65536

rburema, 2021-06-22:
oh true!

Wrap around to discussion 0? Github team sending a cake? Both? I doubt they're 16 bit though :-p

Ghostkeeper, 2021-06-24:
Wait, this is not a new feature! We've had this feature for years.