Ultimaker/Cura #10002: Minimum Support Area affects support models but won't be shown for them if Generate Supports is off

BloodyRain2k, 2021-06-18:
### Application Version

4.9.1

### Platform

Win 7

### Printer

CR-6

### Reproduction steps

I had the Minimum Support Area set to 2 mm^2 for the generated ones.
Then I made a model that needed some better supports than the generated ones and I spent 2 hours tweaking the that model, trying to find out why the lower 2 layers where always missing.

Then I remembered the Minimum Support Area setting, which was NOT shown.
So I turned Generate Supports on, reduced that setting, turned Supports off again, and viola: no more missing layers for my support model.

### Actual results

Please look in the project file, it's a rather specific problem in relation to the models used.

### Expected results

If I add support models while having Generate Supports off, I expect all settings that affect the support models to be shown.

But this one was missing.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Spool_Rail.zip](https://github.com/Ultimaker/Cura/files/6679087/Spool_Rail.zip)

fvrmr, 2021-06-21:
Hi @BloodyRain2k thank you for your bug report.
I will discuss it with the team. Keep you posted!

Ghostkeeper, 2021-06-25:
Indeed, that was wrong. Easy to fix though. Thanks!

commit 914f4bbe (2021-06-25): Show Minimum Support Area setting also when there are only support meshes

The setting influences support meshes. If support was disabled but there was a support mesh, this setting would influence the slice result even if it was invisible.

Fixes #10002.
closed 2021-06-25 by Ghostkeeper