Ultimaker/Cura #9995: Skin Expand distance for Dual extrusion

Jarren141, 2021-06-18:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Custom FFF

### Reproduction steps

Having skin expand distance set under one extruder doesn't take effect unless it is also set for the second extruder even though the 2nd extruder isn't being used to print. Skin edge support distance also has the same problem with it only taking effect if it is set for the second extruder

Looking at layer 376 you can see the Skin edge support not in file 1 and it taking effect in file 2
and at layer 381 is is clear for the skin expand distance not taking effect.

### Actual results

Skin expand distance  and skin edge support not taking effect

### Expected results

Skin expand distance  and skin edge support taking effect

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[New folder.zip](https://github.com/Ultimaker/Cura/files/6673894/New.folder.zip)

fvrmr, 2021-06-18:
Hi @Jarren141 thank you for your report.
Could you also share your printer definition and extruder files? 
I can't open your project file correctly because I don't have your printer in my machine definitions.

Jarren141, 2021-06-18:
I resliced with a default Ultimaker 3 and didn't get the same issue so I probably have something wrong with the Definitions but I don't see that setting overridden in the Ultimaker 3 or its Ultimaker base but I could easily be missing something.
[Def.zip](https://github.com/Ultimaker/Cura/files/6677955/Def.zip)

Ghostkeeper, 2021-06-25:
This is a problem with your extruder definitions. They don't override the `extruder_nr` setting, so both are extruder 0. Since both extruders are sent to the engine as extruder 0, the one that got sent last happens to override the settings of the one before. In your case, the `Essentium_HSE280i_extruder_1.def.json` file, which still has the horizontal expansion set to 0.8mm.

You'll need to add these lines to the extruder definitions. For extruder 0:

    "extruder_nr": {"default_value": 0},

and for extruder 1:

    "extruder_nr": {"default_value": 1},

closed 2021-06-25 by Ghostkeeper

Jarren141, 2021-07-12:
Thanks much! Sorry for taking up your time.