Ultimaker/Cura #9996: Support generation problem

i-R0mka, 2021-06-18:
### Application Version

4.9.1

### Platform

win 10

### Printer

custom

### Reproduction steps

when slicing a model, supports are printed from the second layer

### Actual results

when slicing a model, supports are printed from the second layer

### Expected results

supports should be printed from the first layer

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6675078/cura.log)
[CFFFP_Деталь.zip](https://github.com/Ultimaker/Cura/files/6675083/CFFFP_.zip)
[image]
[image]

fvrmr, 2021-06-18:
Hi @i-R0mka thank you for your report.
Can you share a project file (file - save project). Your 3mf doesn't contain your settings so I can't reproduce your issue.

i-R0mka, 2021-06-18:
[CFFFP_Деталь1.zip](https://github.com/Ultimaker/Cura/files/6675702/CFFFP_.1.zip)

fvrmr, 2021-06-24:
If you disable the setting `brim replaces support` the support is printed from the first layer. 
Or you can disable `brim only on outside` and then the support is also printed on the first layer.

i-R0mka, 2021-06-24:
"brim replaces support" completely disables the printing of support on the first layer, regardless of whether there is a border on this area or not.

Ghostkeeper, 2021-06-25:
The supports are expected to be printed from the second layer if there is brim in its place on the first layer and Brim Replaces Support is enabled.

If Brim Replaces Support is disabled, the support is expected to be dropped all the way down, and the brim should then go around the support as well. This is what I'm seeing. It doesn't seem to disable printing support on the first layer. Here is a screenshot of the first layer of your project when Brim Replaces Support is disabled:

[image]

Are you seeing a different result, then?

i-R0mka, 2021-06-25:
How to get around this bug is clear. This topic was raised precisely because of the illogicality of generating support with these settings. the button description says:
"Enforce brim to be printed around the model even if that space would otherwise be occupied by support. This replaces SOME regions of the first layer of support by brim regions."
In the area where there is no brim, for some reason there is no support. Well, if support is not printed on the first layer, why generate it from the second layer?

Ghostkeeper, 2021-07-01:
Ah, sorry. Yes. The talk about brims led me away from the original bug report. The problem is actually precisely like you describe. The parts of support in the cavities in this print are omitted on the first layer.

The workarounds seem to be to either have brim inside the holes (which is harder to remove) or to make brim override support. Disabling both causes a bug, yes.

I can't reproduce the issue with different models, but with this model it does reproduce. It's expected to print support on the first layer there.

Ghostkeeper, 2021-07-02:
We've discussed this issue with the team and added it to our backlog. Developers, please see issue CURA-8357.

3Duke3, 2021-07-08:
I can confirm I have struck this issue a number of times. To reproduce you need a model with two concentric footprints. It sort of goes hand in hand with models that have an inside brim generated when "Brim Only on Outside" is selected.

[image]

[image]

closed 2024-12-09 by github-actions[bot]