Ultimaker/Cura #10005: Preferences window does not support keyboard navigation

bershanskiy, 2021-06-19:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

N/A

### Reproduction steps

1. Open Cura
2. Via the program menu open "Preferences" window (click "Preferences" > "Configure cura..."
3. Click on "General" to highlight it
4. Press down key and up key to select amy other section of the menu

### Actual results

Nothing happens besides highlight moving from one preference section title to another.

### Expected results

When highlight moves from one preference section title to another, that section is opened. This would be consistent with other programs.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Apparently, there is no way to switch to another section without using a mouse.

fvrmr, 2021-06-21:
Hi @bershanskiy thank you for your report.
I checked versions 4.7 to 4.10 and your expected result doesn't happen in the other versions. So your report is more a feature request instead of a bug. 
I will bring this up with the team. Keep you posted.

Ghostkeeper, 2021-06-28:
I'm unable to reproduce this issue on Ubuntu. Keyboard navigation seems to work fine.
Demonstration: http://dulek.net/work/preferences_keyboard_navigation.webm

By clicking on an element it's clear to see where it's started. If you don't click anything first, pressing tab will first select the header which is not visible. But it'll continue with the real options after pressing tab a few times.

@rburema said he'd take a look on Windows to see if he can solve it, according to our meeting notes.

rburema, 2021-06-29:
@Ghostkeeper That's not what they mean, I think. Go to 'general', then press 'down'; you don't go to the 'settings', or 'printers', or 'materials' menu's. So it's not about the menu's themselves.

I have a partial fix for this, but was stuck on the materials menu, which eats the focus (and the way it's set up, it needs it). So you can go down (or up) to the materials menu with the keyboard, and then it stays on there.

rburema, 2021-07-06:
I made a ticket on our board to review the partial fix. (The partial fix itself can be seen here https://github.com/Ultimaker/Uranium/commit/caf8a3cb775d34bf5108f9d90ee1b96638c802b2 )

Fixing this completely would probably mean untangling the materials menu though, which is unlikely to happen soon.

rburema, 2021-07-09:
I'm sorry, but I've had to revert the partial fix. QA was worried that the behaviour as implemented would be too much like a bug (since getting past the Materials menu isn't obvious).

closed 2024-12-09 by github-actions[bot]