Ultimaker/Cura #10006: SKR Mini 3 V2 USB Print

PistolShrimpPimp, 2021-06-20:
### Application Version

4.9.1

### Platform

Windows 1

### Printer

Ender 3 w/ SKR mini 3 Vs

### Reproduction steps

Install unsigned drivers for SKR 3 mini V2
Plug in Printer
Launch Cura

### Actual results

The software does not communicate with the printer.  I have to remove the USB Printing file just to keep the printer plugged in while I have Cura running or else it completely crashes the printer. When the USB file is removed, I can save to removable (directly to the printer) but none of the printer functions show in Cura (of course).

### Expected results

Real-Time USB printing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't have much more to work from.  This worked on the stock board.  I know this can be driver related but I don't know who else to reach out.  I've attempted already to modify rates/tweak the file based on various online tutorials but it just doesn't seem to work with this board for whatever reason.

GregValiant, 2021-06-20:
Any chance of getting back to BigTreeTech and finding out the Baud Rate?  Cura works from a list to determine the baud by checking them until it finds one that works.  You might need a different driver for the port on the computer as well.

PistolShrimpPimp, 2021-06-20:
Sure I'm happy to look into this to see what can be done. I haven't seen
many reports of this so I would like to think it's something possibly with
my system but I have tried multiple usb controllers and drivers.  I'll
continue to t/s and get back of I'm about to find a solution.

closed 2021-07-09 by no-response[bot]