Ultimaker/Cura #10007: Crashes trying to change nozzle size.

Duke0102, 2021-06-20:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

Pressed the bar to open the drop down to change Material and Nozzle Size,
Pressed Nozzle Size to change the option.

### Actual results

Software closes without error on screen

### Expected results

Expected result was to show option list for changing nozzle size.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Tried running as Admin and also tried 4.9.10 beta.
[cura - Copy.log](https://github.com/Ultimaker/Cura/files/6681972/cura.-.Copy.log)
Thank you.

fvrmr, 2021-06-21:
Hi @Duke0102 thank you for your report. It looks like Cura doesn't have the access to your printer.

```
2021-06-20 09:48:55,040 - DEBUG - [JobQueWorker [2]] USBPrinting.AutoDetectBaudJob.run [49]: Checking COM3 if baud rate 9600 works. Retry nr: 1. Wait timeout: 15
2021-06-20 09:48:55,041 - WARNING - [JobQueWorker [2]] UM.Logger.logException [112]: Exception: Unable to create serial
2021-06-20 09:48:55,043 - WARNING - [JobQueWorker [2]] UM.Logger.logException [116]: Traceback (most recent call last):
2021-06-20 09:48:55,045 - WARNING - [JobQueWorker [2]] UM.Logger.logException [116]:   File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\USBPrinting\AutoDetectBaudJob.py", line 54, in run
2021-06-20 09:48:55,047 - WARNING - [JobQueWorker [2]] UM.Logger.logException [116]:     serial = Serial(str(self._serial_port), baud_rate, timeout = read_timeout, writeTimeout = write_timeout)
2021-06-20 09:48:55,048 - WARNING - [JobQueWorker [2]] UM.Logger.logException [116]:   File "serial\serialwin32.py", line 31, in __init__
2021-06-20 09:48:55,050 - WARNING - [JobQueWorker [2]] UM.Logger.logException [116]:   File "serial\serialutil.py", line 240, in __init__
2021-06-20 09:48:55,051 - WARNING - [JobQueWorker [2]] UM.Logger.logException [116]:   File "serial\serialwin32.py", line 62, in open
2021-06-20 09:48:55,053 - WARNING - [JobQueWorker [2]] UM.Logger.logException [116]: serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)
```

Could you try an other USB port, or try without your printer connected?

Duke0102, 2021-06-22:
Thank you for the response @fvrmr 
I don't have the printer connected, I use octoprint for remote printing. I have tried a fresh install removing all plugins etc with Cura but I'll double check there's no obvious hardware issues that could be causing this.

fvrmr, 2021-06-22:
Thank you, let me know! You could try to disable the USB printing plugin to be sure.

Duke0102, 2021-06-24:
@fvmr thanks a bunch. Never printed using my PC direct and not had issues before so not sure how I managed to do this lol.
Works fine now so thanks again.

closed 2021-06-24 by Duke0102