Ultimaker/Cura #10008: Default material list is re-added when updated app.

FSXLiveriesdotcom, 2021-06-20:
### Application Version

4.8.0 up to 4.10.0

### Platform

Windows 7

### Printer

Ender 3v2

### Reproduction steps

Updated between 4.8, 4.9.0, 4.9.1 and 4.10.0_beta

### Actual results

After each update, all stock materials are back on the material list.

### Expected results

Only manually added materials to be on the list after an update. 
Additionally, during installation have it just as an option to download the default list.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The issue occurs to me since my first Cura usage (4.8.0) and stayed with me up until 4.10.0
I apologize if the report isn't very constructive. It's my first one here.

fvrmr, 2021-06-21:
Hi @FSXLiveriesdotcom thank you for your report.
I would see this more as a feature request than a bug report, I will bring this up with the team. 

But a possible work around for you is make a back up from 4.9.1 and restore it in 4.10.

FSXLiveriesdotcom, 2021-06-21:
Hi fvrmr. 
Jaime van Kessel asked me on discord to report it here. He said it should be automatically. 
Now when I think about it, I'm not sure if we understand each other. 

It would be just handy that those default materials (I already deleted) won't be re-added on each update and only custom ones, added by me while using previous version could be copied to new version during the update.

fvrmr, 2021-06-22:
Hi @FSXLiveriesdotcom how did you deleted those material files? 
Because you can't delete them in the material window.

FSXLiveriesdotcom, 2021-06-22:
I removed all .fdm_material files from '\resources\materials'

nallath, 2021-06-22:
Right, in that case I misunderstood what you asked me. Cura simply installs all files included in the installer, which includes the materials. So if you install a new version, you will automatically get those files back again.

Have yo already had a look at the start-up optimizer? It's a plugin that allows materials / profiles to be excluded from startup.

Ghostkeeper, 2021-06-28:
[image]

FSXLiveriesdotcom, 2021-06-28:
> Right, in that case I misunderstood what you asked me. Cura simply installs all files included in the installer, which includes the materials. So if you install a new version, you will automatically [...]

Sorry for late answer. As a matter of fact, I didn't check this plugin before. It looks like it does what I needed. 
Thanks for responds. 
Cheers

closed 2021-06-28 by FSXLiveriesdotcom