Ultimaker/Cura #10012: Add 0.01 mm increment while printing

gooddealonly, 2021-06-21:
### Is your feature request related to a problem?

I use your Cura Slicer 4.91 to send the print file from the computer to the 3D printer. The Jog position in the Cura Slicer monitor screen is disabled while printing. It would be nice to enable the Z-axis adjustment while printing. As is, the user can pause the print and then adjust.  This would allow the user to make the Z-axis adjust while watching the printing. I often adjust the Z-axis while the printer is printing. I can adjust the Z-axis by 0.01 mm increment if I print from the SD card on the printer (Creality CR-10S Pro V2). However, I can only adjust 0.1 mm on the printer when I print from the USB port while printing using the Leveling function on the printer. As is, Cura 4.9.1 allows adjustment only when the printing job is Paused. It would be nice if the Z-axis can be adjusted from the Cura Monitor screen as precise as 0.01 mm while printing. This would allow the user to precisely change the Z-axis while printing. The X-Y axis adjustment should be disabled while printing. 

### Describe the solution you'd like

My request is to allow the Z-axis adjustment at 0.01 mm increment while printing. That is to add a 0.01 mm button. Adjustment, when Paused, is OK but preferably while printing. Disable the X-Y adjustment while printing to prevent click shifting the X-Y axis.

### Describe alternatives you've considered

None

### Affected users and/or printers

Everyone who sends the print file from the computer to the 3D printer and use the Cura monitor screen.

GregValiant, 2021-06-21:
As a user, it is my understanding that since no Ultimaker printers use the USB Printing plugin that it is no longer maintained by the Ultimaker Cura Team.  That has been true for a few years now.
Your option to change the Z during a print would be to send a Gcode command (possibly M206 Z) to adjust the Z.  Moving to RaspPI/Octoprint would be an option.

On a personal level I'm curious how you are able to tell that the Z needs an adjustment of 1/10 the thickness of a human hair?

gooddealonly, 2021-06-21:
I am new to 3D printing and now to using Cura.  Even print over the network, shouldn't it be nice if the user can adjust the Z-axis while printing.  My eye would not be able to see a 0.01 mm change but the printout will show the difference of, say 0.05 mm.  This is especially true for me while the printer is printing the brim, skirt, or raft.  This gives me an optimal Z-axis high.  Also, when I print  (Creality CR-10s Pro V2) from the SD card, there is a 0.01 mm adjustment on my printer  Also when adjusting the initial high I found sometimes, a 0.1 mm change is either too much or not enough.

GregValiant, 2021-06-21:
You can get the same effect by increasing the flow using the Tune control on the printer, or adjusting the leveling screws while the skirt/brim is printing.  Using the leveling adjustments is the easiest for me.

gooddealonly, 2021-06-21:
If you mean the four bed-leveling screws, it would not be feasible while the bed would be moving.  In the Creality CR-10s Pro V2, there is no tune control therefore no flow control.  The only control while printing is Printing Speed and a 0.1 mm Z-axis when printing from the USB port.

Ghostkeeper, 2021-06-28:
There's three reasons why we don't allow jogging control during the print (unless paused):
* Use cases like yours cause the nozzle to stop or slow down to make a 90 degree turn. The user doesn't expect this, but it does cause blips on your print if you adjust the Z position.
* The printer has a buffer of motion commands, usually 8, 16 or 32 commands, which Cura fills with motions to be executed in the future. So there is up to 32 commands waiting in the queue. If Cura adds another motion command to change the Z position, it can take a minute for that command to be executed, especially when printing something slow like the skin on the first layer.
* It's real easy to mess up your print that way.

gooddealonly, 2021-06-28:
Now I understand the command buffering issue, I agree to adjust the Z-axis while printing would not be feasible or advisable.  However, even if I have to pause the printing, to be able to adjust the Z-axis at 0.01 mm accuracy is needed to get the perfect Z-axis height because a 0.1 mm increment is not sufficiently accurate.  Adding the 0.01 mm increment button is very desirable.

GregValiant, 2021-06-28:
Just talkin' here.  I see 5 ways to do this.

1. Adjust the table leveling knobs on the fly.  My Ender 3 Pro is essentially a CR10 writ small with the same adjustors.  I have no trouble doing this and it's easier to make an early correction than to restart the print.  It would last through the end of the print.
2. Pause the print and make an adjustment to the Z.  Not quick, and filament flow on restart is always tricky.  It would be _**overwritten by the next layer change or Zhop**_.
3. Send a command to change the Z on the fly.  Since the printer is working in Absolute mode you would need to send "the current Z + or - the adjustment".  Generally this would involve a human to look at the LCD and see where it is.  It would also be _**over-written at the next layer change or Zhop**_.
4. Send a series of commands to: change to relative mode, adjust the Z by .01 up or down, change back to absolute mode.  Once again, it would be _**over-written by the next layer change or Zhop**_.  Jogs have a downside because Marlin insists that a movement in relative mode involves the Home Offset.  ("G1 F2700 Z" & (JogDistance + HomeOffsetZ)  Since the Home Offset Z is almost always 0 this is pretty easy to do.  Unless you opt for #5.
5.  Adjust the Home Offset on the fly.  This would not result in an actual movement until a layer change or Zhop.  It would carry through the rest of the print.  This would need to reset at the end of the print.  M206 Z0 could be in your end gcode.

I just spin the wheels.  1/16 of a turn is about .05mm and that's close enough for government work.

gooddealonly, 2021-06-28:
My idea of using a 3D printer is that I should be able to print without becoming a 3D printer engineer, technician, or programmer.   I rather using a 3D printer just like using an inkjet printer.  The difference is the inkjet prints out in seconds but the 3D print out in minutes to hours.  Minor adjustment on the 3D printer I thought is therefore reasonable because just the first layer could take minutes if not more.  All the proposals so far should work, but it is simply a workaround and not resolving the problem.   I am a brand new 3D user, other than learning just to use the printer, I have to learn some CAD programs and the Slicer, I rather have the program, such as the Cura Slicer to offer better features to make the printing as painless as possible for the user.  Is there a slicer out there that has or allows 0.01 mm Z-axis adjustment simply click on a button?

fieldOfView, 2021-06-29:
You can add the 0.01 button yourself by adding 0.01 to the list here:
https://github.com/Ultimaker/Cura/blob/4.9/resources/qml/PrinterOutput/ManualPrinterControl.qml#L408

You can normally find that file in `C:\Program Files\Ultimaker Cura 4.9\resources/qml/PrinterOutput/ManualPrinterControl.qml`

gooddealonly, 2021-06-29:
Thanks for the info.  You make it sound so easy.   But, I will have to take a programming class first If I were to add the button myself! Before I sign up for the programming class, I don't even know what language is this!  Also, even if I did it, I may have to do the same thing again each time Cura releases an update.  Today the 4.10.0 version just released.

Ghostkeeper, 2021-07-06:
There's a 6th method by which this could be implemented to add on to Greg's list: Using G92 the current Z coordinate in memory can be adjusted without actually moving the bed. So envisioning the following g-codes, truncated to show only moves relevant to Z position:

```
; [print a number of layers before the pause]
G0 Z10.0 ;Move to 100th layer.
; Pause
; Commands inserted by jogging up +0.01mm
G0 Z10.01 ; Actually moves the print head (or the bed) physically.
G92 Z10.0 ; Reset Z coordinate so the printer still thinks it's at Z10.0
; [Continue printing]
```

This way it won't get reset when the next Z coordinate comes around. This adjusts all of the subsequent layers... Including any layers printed lower, if one-at-a-time mode is activated. So better not move that nozzle too low high up. But the same goes for the bed screws of course.
This method has three problems that I can think of:
* We'd need to parse the g-code and read out the Z coordinates. Can be done (several post-processing scripts do too), but it'd need to be implemented.
* If we apply the change blindly it will also apply when the printer is not printing. This means that the next print might fail if the user jogs the build plate before a print, and then doesn't home before starting the print.
* Users might expect the jogged movements to be reset once the print continues, especially if we also apply this for X and Y. Their print would suddenly continue at an offset.

Without this change, adding a button for 0.01mm won't really make sense since it can't be used to adjust the print while it's printing, as it is.

gooddealonly, 2021-07-11:
Thanks for the info.  Like I said before I am not a programmer and would not be practical for me to learn to program for this.  Maybe I am expecting too much, but I simply want to be able to make a 3D printout without becoming a 3D printer technician, engineer, or programmer.

GregValiant, 2021-07-11:
Well...the printers are electro-mechanical devices.  They require programming to move so at least some partial knowledge of Gcode would seem to be a necessity.  They require maintenance so being a tech becomes a necessity.  Using the LCD you are using Gcode and doing some minor programming.
Cura, Prusaslicer, IdeaMaker, LottMax, MatterContol, et al are fairly complicated programs and a working knowledge of your slicer-of-choice is a necessity.  You program the slicer to generate a gcode.
The little .4 x .2 rectangle under the nozzle involves fluid dynamics, thermodynamics, material science, mechanics, and other engineering sciences.  You don't need a PHD in any of them, but you are going to find out a little about all of them.
I don't believe there will ever be an "Easy" button for 3D printing.  I suppose you could print one.  You would of course have to program it to do what you want it to do and wire it to the printer and interface it with the computer.  Bluetooth would be nice.
Ask anybody who has printed a bunch of spaghetti - "Entropy always increases"  It's thermodynamics at work.

gooddealonly, 2021-07-11:
I am going to use using a car as an example.  I believe a car is much more sophisticated than any 3D printer, more parts computer controls, fuels injection, transmission...etc.  There is never a need for the user/driver of a car to understand how the car work or how to adjust the valve clearance!  There is no need to understand the aerodynamic of anything. I would say printing out a bunch of spaghetti is NOT the fault of the user.  It is the fault of poor software or hardware design.  I first bought a Creality Ender 3 Pro which there is so much assembling and adjustment, I decided I spend too much time fixing it than using it.  I returned if.  Then Creality Ender 3 V2 3D Printer plus the BL Leveling kit and found it still too much trouble.  I returned it also.  Then I bought the Creality CR-10S Pro V2 3D Printer with BL Touch, I can now doing more printing than troubleshooting and fixing things.  I expect the printer will do what it is supposed to do, without the user wasting a lot of time trying and learning a lot of things that have nothing to do with making 3D printed models.  The purpose of a car is to allow me to go from point A to point B without much of my involvement in the WORKING of the car.  Occasional put in the gas and oil, change the filter is reasonable.  But learn to program the onboard computer to make the car run better?  No.  From my readings, I can see a lot of 3D printer users take pride in that they can fix this defect and work around that defect.  Why should these proud people spending any time on fixing things and NOT making 3D models.  When the next 3D printer model comes out, I expect the cycle starts over again!  I rather pay more for a 3D printer, which I did, than to have saved a little money and have a lot of frustrations.

gooddealonly, 2021-07-12:
The Creality Slicer is a modified version of the Cura Slicer.  That was the reason that bought me here.  I do not agree Creality has no support.  I actually received excellent support from Creality (or the U.S. distributor).  The emails that I sent were responded to promptly.  There were at least 10 emails going back and forth with the tech support people.  They even send emails to ask if the issue was resolved after they send me the solution if I did not give them a confirmation that the issue was resolved.   

I know nothing about Ultimaker 3D printers.   This is an Ultimaker site and I expect most visitors here are Ultimaker 3D printer users.   I bought my Creality 3D printer from Amazon.  I just checked on Amazon, there are only 3 Ultimaker models from $2600 to over $6000 which are way out of my budget.  

I have no idea what class or grade my 3D printer belongs to.  Using your "not a town car" analogy, is my Creality CR-10S Pro V2 a low-cost Kia, and the Ultimaker 3D printers are Mercedes or Lexus?  Are you saying had I bought a more expansive 3D printer, such as an Ultimaker model, I would have a much easier time?  Are you guys much happier than I am with your Ultimaker 3D printer?

GregValiant, 2021-07-12:
I like my Ender 3 Pro.  It's my first and only printer.  I would rather it didn't need constant maintenance, but with a monthly program of preventive maintenance it is fine for the hobby and minor repair stuff that I do.
I don't think the automobile analogy fits.  The automobile is an end product.  A print is an end product.  To get to an end product there has to be production tools.  The printer is a production tool and as such it needs to have direction to produce an end product.  Milling machines needed their handles spun and the printer needs settings adjusted.
I also think it is an absolute plus that programs like PrusaSlicer, Lottmax, IdeaMaker, MatterControl, and yes, Cura, are free.  It is another plus for me that Cura can be customized.  I cut my teeth on analog computers and paper tape.  Python has been a chore, but slowly it comes along and I am able to alter things like my definition file and a couple of plugins to more suit my personal needs.
Creality.  I bought a silent board from Creality.  It lasted 4 days.  It took me 4 months, 5 videos, and 21 emails to finally get them to send me a new one.  The power connector on the new one lasted 3 months before it burned up due to high resistance heating because Creality was too cheap to crimp on a ferrule.  It's a good thing I was sitting there to catch the smell.  I won't even go into the cooling fan fiasco other than to mention that the original only lasted 45 days.
The printer works.
It's very accurate.
I watch it like a hawk.

gooddealonly, 2021-07-12:
I buy most of my equipment & toys from Amazon and if it went bad within 30 days, I would return it to Amazon for a refund or exchange.  I would not put up with anything that is defective in the first 30 days.  
There is a reason why the tech support sometimes takes a few "verifications" to make sure the issue is indeed their product.  I once have a dashcam that went bad after a few months. The tech support wants to test this and test that, photos and videos,  to clearly show the webcam was defective.  They would not want to send out a replacement unless they are 100% sure the item is bad.  Eventually, they sent me a replacement. 

To those Ultimaker 3D printer users:  is your Ultimaker 3D printer a Mercedes or a Kia?

Ghostkeeper, 2021-07-16:
I'm hiding the discussion here about comparing 3D printers with cars, as it has no relation to the original problem of using the jogging mechanism as a way to level/space the build plate.