Ultimaker/Cura #10013: infill double up to improve print time

devbro1, 2021-06-21:
### Is your feature request related to a problem?

in most cases, infill lines are printed on top of each other, either completely or partially. it might be possible to combine infill lines from different layers to cut down on travel time.

### Describe the solution you'd like

assuming there was a checkbox under experimental section, that could enable this behavior:
0. assuming layer height of 0.2 and nozzble size of 0.4
1. at layer 5, shell is printed at 0.2, infill is not printed
2. at layer 6, shell is printed at 0.2, infill is printed at 0.4

or
0. assuming layer height is 0.1
1. at layer 5, shell is printed at 0.1, infill is not printed
2. at layer 6, shell is printed at 0.1, infill is not printed
3. at layer 7, shell is printed at 0.1, infill is printed at 0.3

this approach can decrease the time needed to print infill and thus speeding up the overall print time.

### Describe alternatives you've considered

current alternative is to print lower density infill to decrease the time which does affect the weight and durability of the part printed.

### Affected users and/or printers

any model that has a chunky infill can benefit from this, without compromsing the shell.

GregValiant, 2021-06-21:
Set the Infill Layer Thickness to 2X the regular Layer Height.  The infill will print every other layer.  I usually kick the Infill Line Width up to .45 (.4 nozzle) to account for the extra flow required.  The same trick can be used with Support Infill Layer Thickness.
This is best used with infills like Lines or Grid since each layer is directly on top of one another.  It might not work at all with an infill like Gyroid that moves in the XY every layer.

fvrmr, 2021-06-22:
Hi @devbro1 thank you for your request.
Can you let us know if the work around @GregValiant suggested could work for you?

devbro1, 2021-06-23:
Yes the suggested workaround works as exactly i had in mind. Thank you so much for the info :)

closed 2021-06-23 by devbro1