Ultimaker/Cura #10014: layer height should should raise zhop

252zeke, 2021-06-21:
### Application Version

4.5, 4.6, 4.9

### Platform

Windows 10

### Printer

Creality ender 3, creality ender 3 pro, creality cr10 S

### Reproduction steps

Changed the nozzle size up to 8mm. This causes the layer heights to default to .32mm. Then when you enable zhop, it defaults the zhop to .2mm. Then you slice and print the object.

### Actual results

Print starts out fine and after it gets to about 3mm tall causes the print head to run into the print. and cause large offsets on the prints.

### Expected results

I think the thing that what would resolve this is that the zhop is auto adjusted to be .05mm more than the layer height and support layer height

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have about 4 printers that seem to get this issue, and I have looked into adjusting the pulleys and making sure the z axis rod moves freely and is well lubricated. I have also tried different lubricants with not much success.

fvrmr, 2021-06-22:
Hi @252zeke thank you for your report.

The z hops are defined in per printer and are not related to layer height or nozzle size.
Looking into the definition file for Creality that is defined here: https://github.com/Ultimaker/Cura/blob/61b4dd3efcc01d78a8b2e9ae35b948a2d9710425/resources/definitions/creality_base.def.json#L217

We don't maintain these profiles but Pull Request to change this is always welcome.
You can create a formula here, for example:  `"retraction_hop": { layer_height * 2}`

I will label this issue as deferred, since we don't maintain these profiles. Feel free to reach out if you have more questions.

GregValiant, 2021-06-23:
@252zeke The ZHop height is always an absolute number and is always added "relatively" to the current working Z height of a layer.  In your example, a .32 layer height for 15 layers puts the working Z at 4.80mm.  The ZHop height of 0.2 would be added to that and the nozzle would travel at Z=5.00mm before dropping back down to the working Z=4.80mm.  Theoretically, the nozzle should clear the part.  That isn't always true as sometimes there are random pieces of plastic sticking up above the infill or support and they might be taller than .2 above the working Z.

When I enable ZHop I set it to .5mm to provide clearance for the nozzle over any protuberances that might be in the infill or support.  There have been occasions when I set the ZHop height to 1.0mm.  That happens when I recognize that a feather edge of a feature will curl up and be a problem if the nozzle hits it.

You can create a profile for the .8 nozzle that has ZHop enabled and a ZHop height of 1.0 (or whatever you like).  If you find that the nozzle still drags on the print, you may have to re-visit the mechanicals (though I agree it likely isn't mechanical since it involves three different printers).

252zeke, 2021-06-23:
Hi GregValiant, you have a point, I just struggled with this issue, and I figured out how to solve it with going into the more advanced settings in Cura. However it just wasn't very novice friendly and so it seemed like something could be done, that way if a novice comes by and installs a larger nozzle Cura would hopefully make sure the zhop would clear the support better. With a 4mm nozzle the random bits of plastic don't stick up as high to cause a layer shift, however with an 8mm nozzle the pits of plastic seem to stick up higher than usual.

GregValiant, 2021-06-23:
I was guessing that it was the big nozzle that was leaving higher protuberances.  I think this is one of those things that must be learned through experience.  A .2 Zhop is OK for most things.  I don't use Z hop very much but when I do, I really need it.  I post-process my gcode files and remove the Zhops from those layers where Zhop isn't needed.
Next to the Settings Search box is a drop-down for Setting Visibility.  The first thing I do is make sure it is set to "All".  I really only have one "profile" because every print is at least a little different and requires that each setting be stared at and thought about.  It seems like a PITA but it definitely cuts down on mistakes in my print setups.  An extra 10 minutes going through the settings is better than aborting a print after two hours because I notice that I forgot to move the Z seam.

closed 2023-06-09 by github-actions[bot]