Ultimaker/Cura #10015: Crashs in macOS Monterey

seungwook-jung, 2021-06-22:
### Application Version

4.9.0, 4.10.0 beta

### Platform

macOS (12.0 beta, Monterey)

### Printer

Ender3 pro

### Reproduction steps

cura crashes when App starting in macOS Monterey.

I know that it's too early for request this, but I wander do you have any plan for support monterey and it's schedule.

### Actual results

crash

and it's crash report from OS

```
Process:               cura [2480]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.9 (4.9.1)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2021-06-22 11:11:55.1759 +0900
OS Version:            macOS 12.0 (21A5248p)
Report Version:        12
Bridge OS Version:     6.0 (19P50264n)
Anonymous UUID:        A7BADA99-F6CA-AB4F-F2D1-FA68511EF35A

Sleep/Wake UUID:       FB37DBE4-A41C-4928-8CA4-46A4BBB8C010

Time Awake Since Boot: 3000 seconds
Time Since Wake:       836 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Note:        EXC_CORPSE_NOTIFY

VM Region Info: 0x18 is not in any region.  Bytes before following region: 4302913512
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100794000-1009dc000    [ 2336K] r-x/r-x SM=COW  ...ts/MacOS/cura

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff815e94db6 __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff815ec42d7 pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff815ddb204 raise + 26
3   libsystem_platform.dylib      	    0x7ff815ed94ad _sigtramp + 29

Thread 1:
0   libsystem_pthread.dylib       	    0x7ff815ec00b0 start_wqthread + 0

Thread 2:
0   libsystem_pthread.dylib       	    0x7ff815ec00b0 start_wqthread + 0

[... 1619 lines cut ...]
===========                     =======  ======= 
TOTAL                              3.9G     2356 
TOTAL, minus reserved VM space     2.7G     2356 

Model: MacBookPro16,2, BootROM 1715.0.26.0.0 (iBridge: 19.16.10264.5.14,0), 4 processors, Quad-Core Intel Core i5, 2 GHz, 32 GB, SMC 
Graphics: kHW_IntelIrisPlusGraphicsItem, Intel Iris Plus Graphics, spdisplays_builtin
Memory Module: BANK 0/ChannelA-DIMM0, 16 GB, LPDDR4X, 3733 MHz, SK Hynix, -
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, LPDDR4X, 3733 MHz, SK Hynix, -
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x7BF), wl0: May  3 2021 22:43:35 version 9.30.453.0.32.5.71 FWID 01-79e9fe42
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 85.0
Thunderbolt Bus: MacBook Pro, Apple Inc., 85.0

```

fvrmr, 2021-06-22:
Hi @seungwook-jung thank you, I will bring this up with the team. Keep you posted!

evtrados, 2021-06-25:
Hi @seungwook-jung, 
Cura uses Qt v5.15.2, which, according to their website supports macOS versions from High Sierra (10.13) upwards (https://doc.qt.io/qt-5/supported-platforms.html). So, I would not expect Cura to run on a version earlier than that.

billyjacoby, 2021-06-25:
I'm having a very similar crash, weird thing is it has been working fine from the day the beta came out until this morning. Not sure what changed.

I've got an M1, I'd be happy to help if there's anything I can do to help!

kernelkraut, 2021-06-28:
Same issue here. Hasn't worked from the time I upgraded to Monterey. Mac mini M1. The crash is during a different stage of startup each time. The most common failure to launch occurs at "Initializing Active Machine"

Let me know which log files I can upload for you to have a look at.

nallath, 2021-06-28:
> Same issue here. Hasn't worked from the time I upgraded to Monterey. Mac mini M1. The crash is during a different stage of startup each time. The most common failure to launch occurs at "Initializing Active Machine"
> 
> Let me know which log files I can upload for you to have a look at.

Hmm, that might actually be a different issue. Did you check if the issue is also there with the 4.10 beta?

kernelkraut, 2021-06-29:
> Hmm, that might actually be a different issue. Did you check if the issue is also there with the 4.10 beta?

I pulled 4.9 now (Couldn't find 4.10) - That sorted out my startup issue and the app works fine again. Thanks for the suggestion.

Ghostkeeper, 2021-06-29:
This issue needs to wait on support by the Qt platform.

Our current Qt version (LTS) doesn't support MacOS 12: https://doc.qt.io/qt-5.15/supported-platforms.html

The latest Qt version also doesn't support MacOS 12 yet: https://doc.qt.io/qt-6/supported-platforms.html

There is not even a mention of Monterey yet in their public issue tracker. I think it's still in very early stages for the Qt Company.

seungwook-jung, 2021-06-30:
> Hi @seungwook-jung,
> 
> We've not tested Cura on Monterey yet, since no one has this version of the OS installed (it's still in Beta).
> I can upgrade my Big Sur installation at some point and try to identify why this happens. We sure have to support Monterey once it becomes more stable.
> 
> Some additional information: Do you have an M1 or Intel processor?

Hi @evtrados, 
thank you for reply. (I know it's too early for asking for Monterey, lol)

Anyway, I'm useing `Intel processor`

evtrados, 2021-06-30:
Hi @billyjacoby ,

Indeed the M1 platform is not yet officially supported by Qt (https://www.qt.io/blog/qt-on-apple-silicon), which means that Cura also cannot theoretically be run/used without issues on an M1. It seems, however, that Qt have fixed all underlying issues and will be releasing v6.2 that will support the M1s, so once that is done we will probably need to upgrade our Qt version used by Cura to support it as well. It will be take some time until that happens I suppose.

billyjacoby, 2021-07-04:
@evtrados Hmm that's interesting. I can run Cura 4.8 without any issues - is that based on a different Qt version or something?

Ghostkeeper, 2021-07-05:
Cura 4.8 was running on an older Qt version: 5.10. Since 4.9 we're running on Qt version 5.15.

closed 2024-12-09 by github-actions[bot]