Ultimaker/Cura #10016: (4.9.1 and 4.10.0 beta) Cura does not slice correctly when rotating the model by 90 degrees

cupnb, 2021-06-22:
### Application Version

4.9.1 and 4.10.0 beta

### Platform

Windows 10

### Printer

Anycubic i3 Mega

### Reproduction steps

I exported a file from SolidEdge to STL (which, when viewed in the Windows 3D viewer looks fine) and tried to slice it in Cura. I need to rotate the model 90 deg. Also support is turned off. Other than that I used the Draft preset.

### Actual results

The slicing takes a lot of memory and time. In the end the result is not plausible and one layer seems to get filled completely (even outside of the printers limits). See the attached screenshot.

[image]

### Expected results

A nicely sliced model. The slicing works in 4.8.0. The only troubled versions are 4.9.1 and 4.10.0 beta. (and possibly 4.9, I did not test that one.)

Not rotating the model by 90 degrees fixes the issue but makes it impossible to print the desired model. Also the problem does not occur with other files. (at least none that I know of)

The attached project file and log have been generated with 4.9.1

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6692486/cura.log)
[AI3M_magazine_4.zip](https://github.com/Ultimaker/Cura/files/6692495/AI3M_magazine_4.zip)

fvrmr, 2021-06-22:
Hi @cupnb thank you for your bug report.
I will bring this up with the team. Keep you posted!

GregValiant, 2021-06-22:
One of the indicators I use to determine a bad model is if it slices differently when rotated and what you have there is a bad model.  In the image you posted you can see the gigantic extra surface that was generated.  It's what is causing the huge amount of time for slicing.  The attachment is a repaired model.  See what you think.
[AI3M_magazine_4_fixed.zip](https://github.com/Ultimaker/Cura/files/6694690/AI3M_magazine_4_fixed.zip)

fvrmr, 2021-06-22:
Let me know if the fixed model from @GregValiant works, otherwise I will still discuss it with the team!

cupnb, 2021-06-22:
Yes, his fixed model worked. I am sorry for inconveniencing you!
What threw me off was the fact that 4.8.0 worked flawlessly. I'll have to review my settings in solid edge, I don't know why it does that even after changing the dimensions and some substantial changes to the geometry. I'll make sure to educate me further, I did not know of the pitfalls of STL yet. Cheers!

closed 2021-06-22 by cupnb