Ultimaker/Cura #10017: "Fan Speed" options in modifiers mesh for override the current fan setting

damy2000, 2021-06-22:
### Is your feature request related to a problem?

Is not possible to set an override / modifier for fan speed.

It would be extremely helpful to modify "Fan speed" only for a small area (trought a modifier), especially in the case of materials such as ABS / PC and in particular areas subject to overheating or areas where it is important to improve quality such as overhangs or tiny details.

### Describe the solution you'd like

"Fan Speed" options in modifiers for override the current setting...

### Describe alternatives you've considered

No found any alternative, "Overhang Wall" or "Small Feature" they do not have fan options for override fan speed.

### Affected users and/or printers

For all those who use materials sensitive to thermal deformation, such as ASA, ABS, PC, etc. Allowing calibrated cooling of some areas would be a big step forward!

fvrmr, 2021-06-22:
Hi @damy2000 thank you for your feature request.
I will bring it up with the team. Keep you posted!

GregValiant, 2021-06-22:
I was going to write this up.  I thought a post-processor to add a Cooling Profile would be a better solution.
This is the dialog for the one I wrote for my own use (in VB).  It strips out the existing M106/M107's and inserts new M106's at the requested layer(s).
[image]

And then I came across almost the exact same thought for the one in IdeaMaker.  It doesn't appear to be a post processor (but I didn't look at it that close).

[image]

Being able to increase fan speed for certain features, and then to shut the fan off again when the feature is finished is handy for materials that are subject to warpage/lifting.  If the situation arises multiple times - no problem as the user just needs to anticipate the situation.