Ultimaker/Cura #10018: Wrong skirt lines count in preview

Shine753, 2021-06-22:
### Application Version

4.9.1

### Platform

Windows 7

### Printer

Ender 3

### Reproduction steps

Add model
Slice with 10mm distance for skirt lines from object (problem, on screen and in real life printing)
Slice with 20mm distance for skirt lines from object (no problem)

Seems to happen on small objects.

### Actual results

With 10mm distance, asking for 2 skirt lines, 3 skirt lines are calculated and shown and printed.
With same parameters and 20mm distance, no problem.

### Expected results

Asking for 2 skirt lines should give 2 skirt lines, not 3

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Bug.zip](https://github.com/Ultimaker/Cura/files/6693490/Bug.zip)

Asterchades, 2021-06-22:
Check your Skirt/Brim Minimum Length. This will override the number of lines specified in the event that simply doesn't extrude enough material, which becomes more and more likely the smaller a model is.

Shine753, 2021-06-22:
@Asterchades : correct. Was hidden by default.

closed 2021-06-22 by Shine753