Ultimaker/Cura #10019: 16 extruders makes model transformation controls inaccessible

jenniferlee1818, 2021-06-22:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

custom

### Reproduction steps

1. added custom printer with 16 extruders
2. activate custom printer
3. load stl

### Actual results

the extruders on the left side of GUI cover/push up some of the model transformation controls, making them inaccessible

[image]

### Expected results

should have access to all transformation controls

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[custom3MF.zip](https://github.com/Ultimaker/Cura/files/6694951/custom3MF.zip)

fieldOfView, 2021-06-22:
As a workaround, the Sidebar GUI plugin aligns the toolbar to the top of the window, so the transform toolbuttons will stay accessible. You will still loose a couple of the extruder buttons towards the bottom, but hey, workaround.

jenniferlee1818, 2021-06-22:
OK I will try that.  thanks

closed 2021-06-22 by jenniferlee1818