Ultimaker/Cura #10024: PETG Marketplace

JuFari, 2021-06-23:
### Application Version

4.8.0

### Platform

Windows 10

### Printer

US5

### Reproduction steps

1

### Actual results

I would like to add PTEG in  Cura with the Makertplace but it's not available .
Could you do something?

### Expected results

PETG in Marketplace

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

@

fieldOfView, 2021-06-23:
Unless I misunderstand your problem, it is fixed in Cura 4.9.1 (and newer).
Cura 4.9.1 comes with PETG profiles for the Ultimaker S5, Cura 4.8 does not.

fvrmr, 2021-06-24:
Hi @JuFari thank you for your report.
Like @fieldOfView mentioned above if you want the Ultimaker PETG profiles they are not in 4.8 so you need to upgrade to 4.9.1

Or do you want to add your PETG profiles to Cura?

closed 2021-07-08 by no-response[bot]