Ultimaker/Cura #10025: Accessing Downloads Folder Crashes Cura (4.9.1)

brynnflynn, 2021-06-23:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality CR10s

### Reproduction steps

I upgraded from 4.8 to 4.9.1. I then went to my Downloads folder to load an STL. I discovered the STL was not present, and had been deleted. I went to my Recycle Bin outside of Cura and drag/dropped the file into the Open File dialog window in Cura. Cura crashed. Now whenever I try to open Cura, it opens OK, but then crashes as soon as I open the file picker dialog.

[Cura Crash.zip](https://github.com/Ultimaker/Cura/files/6703518/Cura.Crash.zip)

### Actual results

Cura hard crashes to the desktop.

### Expected results

Cura should allow me to at least browse to a different location.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6703512/cura.log)

fvrmr, 2021-06-24:
Hi @brynnflynn thank you for your report.
Do you maybe have Chitubox installed?

brynnflynn, 2021-06-24:
> Hi @brynnflynn thank you for your report.
> Do you maybe have Chitubox installed?

I do, actually. I didn't have this issue with 4.8.

fvrmr, 2021-06-24:
Than your issue is a duplicate of https://github.com/Ultimaker/Cura/issues/9668
There you can also find why this is happening and how to fix it: https://github.com/Ultimaker/Cura/issues/9668#issuecomment-828992971

I will close this issue as a duplicate, but if you have more questions please ask them in the other thread so we keep everything in one place.

closed 2021-06-24 by fvrmr