Ultimaker/Cura #10026: Custom Support screws up view of object

Lumanet2012, 2021-06-23:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Hi everyone,

I am having to print a modified housing for the HERO 5 and I do need to do custom support with 10mm size. 
Below the link show's the screenshot of before and after I applied 1 custom support object. 

https://owncloud.lumanet.info/s/mboNjAOsUCrUPaw

Somehow when I apply the first custom support object which is a square support everything become a dark gray screen and I no longer can see the object only part of the highlighted object. I am not sure what causes this but it's bugging me because I don't do my job without it working proper. 

### Actual results

I do only see a small outline of my part and nothing else not even the build plate is visable. 

### Expected results

I should see my full model with the custom support added but I don't.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://owncloud.lumanet.info/s/mboNjAOsUCrUPaw

fvrmr, 2021-06-24:
Hi @Lumanet2012 thank you for your report.
Does it only happen when you add custom support on every model? Or just this model?
Could you maybe share your project file. I can't really see what is happening from the screenshots.

Lumanet2012, 2021-06-24:
it happens on all models. I just tested it and verified it.

fvrmr, 2021-06-25:
Can you share your log files and project file (file - save project)? 
You can find your logs here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Lumanet2012, 2021-06-25:
I uploaded them. You can find it all in the link I have been shared before.

Ghostkeeper, 2021-06-30:
This issue reproduces with the given project file. Looks a bit different for me, but similar in nature:

[image]

Note also in the bottom left that Cura indicates that the print is 1.822*10^12km wide. Like, 12000 times the distance from the Earth to the sun. And 60mm tall. Cura indicates that the scale is set to 100%. The big model is set to be printed as support. So it seems that the Cylindric Custom Support plug-in generated a model that is unrealistically huge. 3 of them, in fact. Maybe you tried to place 3 custom support objects?

This bug seems to be reported as well by other users, in the issue tracker of that plug-in: https://github.com/5axes/CustomSupportCylinder/issues/7

They are looking for better reproduction steps there, so I hope you can add your experiences there to narrow down the bug in that plug-in.
It is not a Cura bug as far as I can see, but a bug in the plug-in, so I'll close it here.

closed 2021-06-30 by Ghostkeeper

5axes, 2021-06-30:
And for me there is no issue with the Plugin ( tested with Cura 4.10) 
[image]

So .... Just say good luck

Ghostkeeper, 2021-07-01:
Yeah when I tried placing some support objects with the plug-in it also worked fine. But somehow it triggered for Lumanet2012.