Ultimaker/Cura #10027: Different menus with different language settings

Beckerb, 2021-06-24:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Bibo2

### Reproduction steps

i searched for the setting for "retreat distance" in the german gui, and could only find "retreat speed".
the value was 20 mm/s
I then switched cura to english and found the distance parameter. But the speed here was now set to 25 mm/s.
Back in German, the 25 mm/s remained.

### Actual results

nothing - because i just need this information for support issue

### Expected results

same parameters in all languages

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura_.pdf](https://github.com/Ultimaker/Cura/files/6706970/cura_.pdf)

fvrmr, 2021-06-24:
Hi @Beckerb thank you for your report.
I checked the German and English interface. 
[image]

[image]

And they look the same to me, only combing mode is not visible in the German interface. 
For me switching from languages didn't change the values. As you can see I changed the retraction speed to 20mm/s.

Maybe your settings visibility isset to basic or custom. Let me know if this helps for you!

Ghostkeeper, 2021-07-01:
The settings visibility should not affect the search results, if searching, though. Only if the search bar is empty.

What text was Beckerb searching for in the German screenshot? It's cropped off. If it's "retreat distance" I'd expect you wouldn't get any results since that is English, and Cura is brave enough to never retreat.

Beckerb, 2021-07-01:
hello
it's not a matter of "search", the menue's have different entries - in english more than in german.
currently I don't have access to Cura, but the screenshot actually shows it, entries that are present in the english menu are missing in the german one

Ghostkeeper, 2021-07-01:
Your screenshots show settings lists where different settings are shown. That could be because you're opening different versions of Cura, because you have different settings visible, different setting values, different visibility profiles or different search strings. But I'm not able to reproduce that with your reproduction steps so far. Could you further specify your reproduction steps?

These are the reproduction steps I'm taking now:

1. Switch Cura's language to German and restart.
2. Look into the Bewegungen category for the setting "Einzugsabstand".
3. Switch Cura's language to English and restart.
4. Look into the Travel category for the setting "Retraction Distance".

You claim that Einzugsabstand is not present, but Retraction Distance is. That is not what I'm seeing.
German:
[image]

English:
[image]

This looks identical to me. Setting values, which of them are visible, all the same.

We'll need to have better reproduction steps to be able to reproduce this issue. Maybe it would help if you created a numbered list of steps, rather than just a story? Like, step 1... step 2... etc.

Beckerb, 2021-07-05:
Sorry fort he late answer…

Now it seems to work fine – curios..

Thanks a lot

Best regards

Bernhard

[Screenshot (200)]<https://user-images.githubusercontent.com/77000924/123284875-e304d280-d50c-11eb-9ba5-ccb7cc86369a.png>

[Screenshot (201)]<https://user-images.githubusercontent.com/77000924/123284881-e304d280-d50c-11eb-994b-583286bd186b.png>

fvrmr, 2021-07-05:
Hi @Beckerb thank you, I will close this issue and mark it as solved!

closed 2021-07-05 by fvrmr