Ultimaker/Cura #10028: Using "Modify settings for overlap" does not affect wallcount anymore

BloodyRain2k, 2021-06-24:
### Application Version

4.9.1

### Platform

Win 7 x64

### Printer

CR-6

### Reproduction steps

Place a model over another and set the overlapping one to "Meshtype: Modify settings for overlap" and leave the wallcount at 0 as per default

### Actual results

Wallcount unaffected, modifying infill still works though.

### Expected results

Being able to modify the wallcount, in this case to speed up the prototyping by cutting off the model early (that's also why part of it is in the "ground").

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

[Spool_Roller_Clip.zip](https://github.com/Ultimaker/Cura/files/6708107/Spool_Roller_Clip.zip)

BloodyRain2k, 2021-06-24:
Ok, I found out that I can get the whole thing to work with the Cutting mesh mode.

Nevermind. Although I did try that too, I have no idea why it didn't work before and now did...

closed 2021-06-24 by BloodyRain2k

Ghostkeeper, 2021-07-01:
It also works as an infill mesh, like in your project file. Just set the number of walls in the infill mesh to 1 to get 1 extra wall.