Ultimaker/Cura #10029: Bridge Mode assigns wrong flow values for top/bottom layers after the "Bridge" has been done

BloodyRain2k, 2021-06-24:
### Application Version

4.9.1 and 4.10-Beta

### Platform

Win 7 x64

### Printer

CR-6

### Reproduction steps

Have a model that somewhere causes bridge mode to do "it's thing", I guess a cube could already be enough, but I've attached my model to be sure.
Set the 1st bridge layer flow to something below 50%.
Get awful top layers because even after the 3rd bridge layer, the flow setting is not being reset.

### Actual results

Exaggerated settings to show that they mess even with the last top layer:
[image]

Changed the settings to prove that it's the bridge mode settings
[image]

I even tried to change the top and bottom layers to 15 to make sure it wasn't suddenly being fine afterwards (I rarely print with more than 3-5 top/bottom layers), but nah, it kept being like that.
On top of messing with the top/bottom layer settings after it should be done, it's not even ever touching the 2nd and 3rd layer settings even with "Multiple bridge layers" enabled.

### Expected results

Whatever the proper result for bridge mode would be...

Certainly NOT messing up my top layers. I've been already wondering for a while why my top layers started looking so bad for a while now, but I didn't expect this to be the cause.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Spool_Roller_Clip.zip](https://github.com/Ultimaker/Cura/files/6708692/Spool_Roller_Clip.zip)

fvrmr, 2021-06-25:
Hi @BloodyRain2k thank you for your report. 
I am not sure if I understand your report correctly, do you mean those thin lines that are on the top layer?

BloodyRain2k, 2021-06-25:
@fvrmr no, the many thin lines are from Ironing. The lines I'm talking about are the ones that are going in triangles on the 1st screenshot and parallel in the 2nd.

Ghostkeeper, 2021-07-01:
The issue here arises from the Bridge Skin Support Threshold being set to 100%. This setting basically says that any skin part supported under X% or less of its area needs to use the bridging routine. Since you've set it to 100%, that means that any skin part supported under 100% or less of its area needs to use the bridging routine. So... all skin uses the bridging routine.

The lowest of the top skin layers is 100% supported by infill. The higher 4 layers are supported by skin. But as it is, setting that to 100% causes the first-layer bridging to apply to all skin.

I suggest you set the Bridge Skin Support Threshold to 99.9% if you want to prevent this case, or to revert it to the default of 50% to cause it to bridge only parts that should reasonably be bridged.

If you want to approximate this behaviour without losing thickness of skin, I suggest you look into the Gradual Infill setting. It does something similar, but in the infill rather than sacrificing skin.

closed 2021-07-01 by Ghostkeeper

BloodyRain2k, 2021-07-01:
@Ghostkeeper ohhh, I see. I indeed set the Threshold to 100% at one point, because I was trying to figure out what it does.
But after I gave up on that I just left it at 100%.

I'll try that an see if that's the whole for my issue. Thanks.