Ultimaker/Cura #10032: missing layer parts

Reksterlord, 2021-06-24:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ender 5 pro

### Reproduction steps

I opend a STL file and looked at it and scaled the model size to 35%, the model was OK. 
Then I sliced the model.
[image]

### Actual results

When I sliced the model and saw the 'PREVIEW' I noticed some gaps in the model after I sliced it.
After printing the model, I saw the same gaps on the print.
[image]

### Expected results

These should't be any gaps 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fvrmr, 2021-06-25:
Hi @Reksterlord thank you for your report.
Could you share your project file (file - save project)?

Reksterlord, 2021-06-25:
[VB4.MD](https://github.com/Ultimaker/Cura/files/6714971/VB4.MD)

fieldOfView, 2021-06-25:
That is an STL, not a project fille. The STL file has just the geometry, but a project file would include both the geometry _and_ the settings you currently use. With a project file we can see what is going wrong. With the STL we can only admire the model.

Anon0ne, 2021-06-29:
Having this issue also... When I rotate the part it will place gaps in different location. Just updated to 4.10 today.
[image]
[CE5_watchstand_newV.zip](https://github.com/Ultimaker/Cura/files/6736709/CE5_watchstand_newV.zip)

fieldOfView, 2021-06-30:
@Anon0ne in your case it looks like your problem is that Coasting is turned on. Turn it off and you gaps will go away.

Ghostkeeper, 2021-07-01:
To create a project file, you'd go to File -> Save Project in Cura. The result is a .3mf file with extra information on how to slice the object.

closed 2021-07-09 by no-response[bot]