Ultimaker/Cura #10035: Tree support issue - missing layers

Eudaimonium, 2021-06-25:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Load this file: 
https://www.thingiverse.com/thing:2116861 

Increase scale to 140 or 150%, use Tree support, use Brim, Slice. Try rotating the model few different ways as well. 

Observe that sometimes, pieces of the tree support will produce a brim outline, but will not actually print until sometime in the upper layers. At some point the tree will appear but it will basically try to print "mid-air", because lower layers are missing. 

[image]

[image]

[image]

[image]

In the attached project file, layer 140 displays the problem very nicely, missing supports under the tail. 

### Actual results

View images attached to repro steps 

### Expected results

The tree support is continuous and non-interrupted. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Mewtwo_20161005.zip](https://github.com/Ultimaker/Cura/files/6717908/Mewtwo_20161005.zip)

GregValiant, 2021-06-25:
I'm not getting that no matter how it's oriented.  Use File | Save Project and post the project file here.
[image]

Eudaimonium, 2021-06-25:
An example project file is included in the post, the last link under "Additional files and info".

GregValiant, 2021-06-26:
All I see is an STL file.  A project file contains the model, and your printer, and your Cura settings and has a "3mf" file name extension.  You might have to Zip it to post it here.

Eudaimonium, 2021-06-26:
I have, of course, zipped the wrong file. My apologies. 

A correct file is here: 
[CE3_Mewtwo_20161005 - Copy.zip](https://github.com/Ultimaker/Cura/files/6719671/CE3_Mewtwo_20161005.-.Copy.zip)

I have discovered something that might be related. With the "Support Wall Line Count" set to zero, it actually has 1 wall, which is sometimes missing. 

You can see this in this project, layer 21: 
[image]

Setting Support Wall Line Count to 1, actually means two walls and then they are always there: 
[image]

GregValiant, 2021-06-26:
Within the 3mf file you have Support Wall Count = 0, Support Density = 1%, and Connect Support ZigZags is unchecked.  You are providing very little support for the support.  Once the branches start to grow you have not provided them with a trunk to grow from.  In addition, that model has long thin areas of support that will need more plastic or they will fail.  Without an outer wall around the support all you have is the ZigZag lines at 1% density.  What you are seeing is what you told Cura to do and that is to build support with no plastic.

My suggestion is to set Support Wall Count = 1 to provide a periphery, Support Density = 20%, to provide a decent base to build the Support Roofs on, and check Connect Support ZigZags to limit retractions.  Also consider increasing the support roof density to 60% to give a better base for the fine features to build on.  I would move the model up off the build plate 1mm so the whole thing builds on support.  Those extremely small spots that the model starts with on the build plate don't have enough area to stick well and could cause early failure of the print.

Ghostkeeper, 2021-07-02:
The "Zigzag" pattern happens to follow the border of the support area for over half of the border's circumference. But by the nature of zig-zagging, the line is going to be on one side for a while, and then on the other.

When the tree support splits into two pieces at layer 21, under Mewtwo's hip, suddenly there are more sides to both pieces and it happens to end up zig-zagging on the sides that previously didn't exist, causing that zig-zag to overhang.

Normally that sort of thing would not be overhanging for very long, just for one side of the model between two adjacent infill lines. But since you've set the infill density to 1%, and since the tree support is fairly "fractal", that can become quite long.

I agree with GregValiant's advice of Support Wall Count to 1 (that is the default I chose myself when I developed tree support) and would even say that you could consider setting the density to 0% as well to just rely on the extra support wall.

This is expected behaviour though. Just weird settings, but not a bug since all the settings do what they are asked. So we'll close this issue.

closed 2021-07-02 by Ghostkeeper

Eudaimonium, 2021-07-02:
I didn't know that this is intended behavior. Thank both of you for your advice on tweaking the settings - I'm still fairly new at this, so this is much appreciated.