Ultimaker/Cura #10036: Infill lines visible on the skin

lorak41, 2021-06-26:
### Application Version

4.9.1

### Platform

macos

### Printer

Cr10s

### Reproduction steps

Print using Printer profiles for e.g Cr10 or CR-X printers. 
I tried using 0% overlap and 3 perimeters (width: 0.4mm), did not help.

### Actual results

I can't seem to find why when I use newest Cura I have these infill lines on the skin. I don't see it with PrusaSlicer using the same printer.

I set infill overlap to 0% and that did not help.

### Expected results

The infill lines should not be visible.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2021-06-27:
What is the wall thickness?  (I use three loops as a minimum).  This can also happen do to under-extrusion.  Have you calibrated the E-steps?  If it is like my Ender it was delivered with E-steps at 93 but after calibrating I'm at 103.  It was delivered under-extruding by 10%.
You may have some mechanical wobble or looseness going on there as well.  Now don't get mad - but none of those prints really look all that good.  I agree the Cura samples are a D and a D-, but the PrusaSlicer print I can only give a D+.

lorak41, 2021-06-27:
I agree with you, they all look pretty bad. I try to eliminate the problems. 
I used the same flow rate for Cura and PrusaSlicer. Thank for the suggestions, I will calibrate esteps.

lorak41, 2021-06-28:
[image]
This is with better filament, 3 perimeters (0.45mm line width), 0% overlap and still, I can see those infil lines. ;//

GregValiant, 2021-06-28:
Part of "seeing the infill" is the filament color.  Translucents are tough to hide the infill and dark colors are the easiest.  I guess it depends on what the manufacturer used for colorant.  The minimum number of wall loops is generally accepted to be 3 but for light and, translucent colors it will be 4 and maybe even 5.  Minimum Top/Bottom is 4 and the same "rules of color" would apply.
With the overlap at 0% the part won't be as strong because the infill isn't bonding to the walls.  You can try "alternate extra wall" to bond the infill to the walls better.

lorak41, 2021-06-28:
Thank you for the suggestions.

I get your "filament color" argument, but how to explain the fact that the same filament does not show infill with Prusa slicer. Moreover, in Prusa the default overlap is 25%, whereas in Cura I set 0% (following the profile of Ultimaker S3, they set 0 there as well).

GregValiant, 2021-06-28:
I can't explain it.  I tend to use Ideamaker more than I do Prusaslicer.  They all have their strenths and weaknesses.  I've been collecting slicers and I'm up to 17.  Most are passable, a couple really suck, Cura (both the main branch and the smartavionics branch) and Prusaslicer are on top.
Have you tried measuing the wall thickness of the different cubes?  Maybe print a couple with no roof so you can get the calipers in.  It might show something.

3Duke3, 2021-06-30:
Does Prusa slicer print infill before walls, or walls before infill?

In cura you can turn off "Infill before walls" - usually enabled by default and that should give you a better result.

lorak41, 2021-06-30:
I've just checked gcodes, you are right, Prusa prints the wall first. Thank you so much!

3Duke3, 2021-06-30:
no probs, I dont know why infill before walls is enabled by default, it should really be the otherway around. It can be usefull in some cases when you have difficult overhangs, as the the perimeters can grab onto the infill that already been printed, but it does require extra tuning to reduce the infill visibility in the wall. I do believe you can set the infill overlap to a negative value. 

Or you can also try, enabling "infill before walls", but also enable "outer before inner walls", you will need to have a wall count greater than 1, but it should print infill first, then print the outer wall second, then print the inner walls third which would in turn connect the outer wall and infill together.

Ghostkeeper, 2021-07-05:
The infill is printed before walls by default because then the walls have something to hang on to on the new layer, if it is next to infill. This reduces sagging, producing stronger and more accurate parts (at the cost of visual quality).

We'll close this because it is a print quality issue dependent on settings. If you think the profiles for your printer need to change, a pull request is welcome, but we won't change these ourselves because we can't test the impact on your printers. Intuition and experience only goes so far.

closed 2021-07-05 by Ghostkeeper