Ultimaker/Cura #10039: slicing process new ends with tree support option.

Ari-SSO, 2021-06-26:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Custom Printer

### Reproduction steps

1. select tre support in a complex figure, and never end

### Actual results

never end the process.

### Expected results

end the slicing process with tree support

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_1.zip](https://github.com/Ultimaker/Cura/files/6720770/CFFFP_1.zip)

I upload the project file.

fvrmr, 2021-06-28:
Hi @Ari-SSO thank you for your report.
I can reproduce your issue with your model. When I changed the support pattern to lines or zigzag it slices perfectly.
When the support pattern is set to cross and changed the support line distance to 1 mm is also slices.

I will discuss this issue with the team. Keep you posted!

fvrmr, 2021-07-05:
I have created a ticket on our backlog for this issue. Devs see CURA-8361

ThomasRahm, 2021-11-12:
I think this is a duplicate of https://github.com/Ultimaker/CuraEngine/issues/1412
If the support density is set to something above 0% it will work.

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

ThomasRahm, 2024-11-22:
Can be closed. Was fixed with https://github.com/Ultimaker/CuraEngine/commit/a0b20d8fbe29913b93b4410212471b2847af8ea3

closed 2024-11-22 by GregValiant