Ultimaker/Cura #10040: Cura is unable to generate normal Bridges

s777s, 2021-06-27:
### Application Version

4.9.1

### Platform

Linux

### Printer

Anet A8

### Reproduction steps

1. Enable Experimental -> Enable Bridge Settings
2. Use the default Settings
3. Load my model  (like a box. about 200mm long; 20mm width inside and 24mm outside)
4. Slice it

### Actual results

The result is definitely unprintable model with too long 200mm bridges in full length of long model.

### Expected results

It must be fine printable model with perpendicular short 20mm bridges.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There is no setting to "shorten bridges (if possible)" or better the default behavior to slice model with optimal bridges length.

smartavionics, 2021-06-27:
Hi, could you please provide that model file? You'll probably need to zip it to get github to allow it to be attached. Thanks.

GregValiant, 2021-06-27:
You might try changing your Top/Bottom Line Directions to [135,45].  Because your model is at 45° you may need to rotate the line pattern.

fvrmr, 2021-06-28:
A project file (file - save project) would be very help full like @smartavionics mentioned please attach it as a .zip here.

closed 2021-07-12 by no-response[bot]