Ultimaker/Cura #10041: Editing Filament Cost Impacts Material Information Globally

Fyrwurm, 2021-06-27:
### Application Version

4.9.1

### Platform

Windows 10

### Printer

Creality Ender 5 Pro

### Reproduction steps

1. Duplicate Generic PLA Material
2. Configure new PLA Information (specifically filament cost)
3. Duplicate Generic PLA Material
4. Configure new PLA Information (specifically filament cost) for additional material

### Actual results

Editing the Filament Cost for the second material altered the Cost for all other created materials

### Expected results

In the same way that Print settings are unique to each material, I would expect to be able to set different costs for different filaments.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2021-06-27:
Scroll down on the right, unlink the material.

Fyrwurm, 2021-06-27:
Thank you for the recommendation, but I do not see any options to unlink.  Can you direct me where to find it?

[image]

fieldOfView, 2021-06-27:
The "unlink" button is only shown on materials that are linked to other materials.

If you create a new material by Duplicating an existing material, they will be linked. The cost and weight of "linked" materials is linked. If you unlink them, they will no longer be linked. Cost and weight of materials that are not "linked" should not affect eachother.

Fyrwurm, 2021-06-27:
Ok, I have wiped out all materials that I created and repeated the steps from my initially reported issue.  The "Unlink Material" button is now appearing.  I have no idea why it was missing when I originally attempted these steps.  This may be a one off error (I have experienced some random program crashes since updating from 4.8 to 4.9.1).

Thanks for the help!

[image]

closed 2021-06-27 by Fyrwurm