Ultimaker/Cura #10043: Inital Layer options for coasting and adaptive layers

FountainGoat, 2021-06-28:
### Is your feature request related to a problem?

When using coasting or adaptive layers for printing thin objects the initial layer adhesion fails more frequently due to less material being extruded.

### Describe the solution you'd like

Adding an option to disable adaptive layers and coasting for the initial layer. This will allow the initial layer to have the best chance of adhesion while the rest of the print can have the quality benefits of coasting and adaptive layers.

### Describe alternatives you've considered

Seperate options for initial layer for adaptive/coating but since this is only 1 layer high its necessary.

### Affected users and/or printers

Everyone using experimental features

fvrmr, 2021-06-29:
Hi @FountainGoat thank you for your feature request.
I will bring this up for discussion with the team. Keep you posted!

Ghostkeeper, 2021-07-06:
Adaptive layers is already disabled for the first layer. The first layer should always be printed at the Initial Layer Height.

Are you seeing that differently? Do you have a project file where the first layer is not printed at the initial layer height?

FountainGoat, 2021-07-06:
I did have a project that printed the initial layer too thin while adaptive layers was enabled, but i am not able to reproduce it now unfortunately.

Ghostkeeper, 2021-07-07:
Hmm, okay well that would still be a bug, if we can find a way to reproduce it...

FountainGoat, 2021-08-12:
I have encountered the bug again and have attached my project file and screenshot. I think it only happens when the model is moved below the bed plane and sliced.
[coastingbug.zip](https://github.com/Ultimaker/Cura/files/6974081/coastingbug.zip)
[image]

closed 2021-08-12 by nallath