Ultimaker/Cura #10045: cura closing

Cleyton-Faria, 2021-06-28:
### Application Version

Arachne engine beta

### Platform

windowns 10

### Printer

longer lk4 with robin nano V3

### Reproduction steps

sometimes the software closes after ejecting the media or after updating the STL

### Actual results

.

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

fvrmr, 2021-06-29:
Hi @Cleyton-Faria thank you for your report.
Could you share your log file? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

fieldOfView, 2021-06-29:
This is a duplicate of quite a few reports on the Arachne releases. Logs aren't really necessary at this point (though be sure to include them next time you open an issue!)

fvrmr, 2021-07-02:
It is indeed then a duplicate of https://github.com/Ultimaker/Cura/issues/9057 it was not really related to Arachne, also happened in 4.8 but it is fixed and will not be an issue for next releases.
Closing this because it is a duplicate.

closed 2021-07-02 by fvrmr