Ultimaker/Cura #10053: brim is printing on inside even if you check brim only on there outside only

jonnyboytech, 2021-06-29:
### Application Version

4.9.0

### Platform

window 10 pro

### Printer

ender 3 pro 

### Reproduction steps

the material I'm using have can lift on the conors so I need a brim to prevent that but my design have my logo on the frist 2 layers and brim still show there and print on areas that (brim only on outside) show prevent brim from print on the inside of the design 
[image]

### Actual results

prints just fine but is a problem with brim setting if you want to add another color or another print first to have a logo with a different color
[image]
 
the material needs to be used with a bring so it don't lift on the conors 

### Expected results

some designs can't be printed with thoes changes are no made in the program

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

just letting you guys know that way  the amazing program you guys created gets even better

3Duke3, 2021-06-30:
This is to do with the inner "D" shapes inside the letter "B" touching the build plate, for what ever reason the logic determines them to have outside brim geometry. 

I would love to see a proper solution for this but imagine it may be hard to implement in cura.

As a workaround, you could try exporting the internal D shapes as separate models, and turn the brim off under per model settings?

fvrmr, 2021-07-01:
Hi @jonnyboytech thank you for your report.
Like @3Duke3 mentioned above this is correct. We have a issue related to this: https://github.com/Ultimaker/Cura/issues/4222 and this is linked to a ticket on our backlog. 
So I will close this issue a duplicate. Hope you understand!

closed 2021-07-01 by fvrmr