Ultimaker/Cura #10056: Cura Crashing

Arcaver, 2021-06-30:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creailty CR-6 SE

### Reproduction steps

1. Load up Cura
2. Click folder to open file
3. Try to select any file to open

### Actual results

Cura will freeze and crash.

### Expected results

Cura will freeze and crash

[cura.log](https://github.com/Ultimaker/Cura/files/6737501/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The only way I can open a file in Cura is through File Explorer. I have to locate the file, right click the file and then select Open With Cura. Once I have the file sliced and saved I have to delete the current file, re-open File Explorer, and then drag and drop the file into Cura in order to work on another project.

3Duke3, 2021-06-30:
I too have this issue, clicking on the folder icon to import a model causes Cura to crash. drag and drop a stl file is ok.

Do you also have Chitubox slicer installed?

I believe there is an incompatibility issue somewhere as when I uninstall Chitubox the problem is resolved. Chitubox included some sort of stl thumbnail preview, and I suspect this is causing the issue, but disabling the thumbnail preview doesn't seem to make any change.

Arcaver, 2021-06-30:
> I too have this issue, clicking on the folder icon to import a model causes Cura to crash. drag and drop a stl file is ok.
>
> Do you also have Chitubox slicer installed?
>
> I believe there is an [...]

I do still have it. I need it for my resin printer so I can't get rid of it.

3Duke3, 2021-07-01:
> I do still have it. I need it for my resin printer so I can't get rid of it.

Same here, it's infuriating. Although I don't think Chitu will ever do anything about it, well, because they are absolutely useless!

Chitubox also caused issues with Fusion360 when exporting stl files, F360 would crash as soon as the file browser opened to chose a save location. Fusion has however managed to resolve the issue, so perhaps Cura can also provide a solution.

fvrmr, 2021-07-01:
Hi @Arcaver thank you for your bug report. 
Like @3Duke3 mentioned this is indeed caused by Chitubox. There is not really something we can do unfortunately.
Here you can find the duplicate of this issue: https://github.com/Ultimaker/Cura/issues/9668 and the only thing you can try is to uninstall the `CHITUBOX_Thumbnail.dll` file in the installation directory by following these steps:
1. Open cmd with administrator rights
2. Enter the command: regsvr32 / u "path (CHITUBOX_Thumbnail.dll path)"
3. Restart the resource manager

I will close this issue as a duplicate but if you have more questions please add them to the other thread so we keep everything in one place.

closed 2021-07-01 by fvrmr

3Duke3, 2021-07-03:
Excuse my lack of deeper software understanding, but seeing as Cura is such a major STL application, would there be an option to develop your own STL thumbnail tool that took priority over shitubox?

Ghostkeeper, 2021-07-07:
@3Duke3 Please continue the discussion in the original ticket, not here.