Ultimaker/Cura #10057: Preview mode exploding model

docksteaderluke, 2021-06-30:
### Application Version

4.9.1 and 4.10

### Platform

macOS Big Sur v11.4

### Printer

Creality Ender 5 Pro

### Reproduction steps

1. Upgrade to 4.10

### Actual results

Everything now explodes in Preview mode even after downgrading to 4.9.1.
This issue did not occur prior to update.
[image]

### Expected results

Model shouldn't explode.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Changing slicer settings doesn't change anything.
This happens to _all_ models now. I've tested with both new models and those that I previously confirmed as working before the update.
No crashing occurs.

docksteaderluke, 2021-07-02:
Hi @fvrmr,

No, I do not! Here are my machine's specs.

[image]

Ghostkeeper, 2021-07-07:
You also posted this on Reddit. I recognise the picture. Here is the post: https://www.reddit.com/r/Cura/comments/oazfff/just_upgraded_to_cura_410_and_my_preview_mode_is/

I have to agree with ahoeben there that this is most likely a graphics driver issue. Nothing really changed between 4.9 and 4.10 with how we render layer view (other than the addition of another colour scheme which you're not using here). You reported there that you also upgraded MacOS itself at the same time, which seems like a more likely culprit.

Could you disambiguate by seeing if the same issue also occurred in Cura 4.9 while using your new MacOS version?

So far none of our developers have been able to reproduce this issue. We haven't seen it happen yet.

docksteaderluke, 2021-07-07:
Yes — I had originally posted it on Reddit then thought this might be a more appropriate place.

I've tried a number of things over the last few days and rebooted my MacBook a few times and, strangely enough, the last time I rebooted it seems to have fixed the issue (not sure why it didn't fix it the first couple of restarts) 🤷.

At this point I can no longer reproduce the bug either so I think it's safe to close this issue. I'll report back if the issue pops up again.

Thanks for your help!

closed 2021-07-07 by docksteaderluke

Ghostkeeper, 2021-07-07:
Perhaps there was an update to the drivers in the background then. But it's still good to know of the issue for other people :)