Ultimaker/Cura #10058: Costing Generator

3Duke3, 2021-06-30:
### Is your feature request related to a problem?

No problems, just a feature request.

### Describe the solution you'd like

I would like the ability to enter a per hour $ value/rate specific to each printer, and have the price displayed in the estimated print time box, be a calculation of ("Filament cost" + ('print duration' x 'printer's hourly rate')). 

### Describe alternatives you've considered

Currently we have the ability to input material cost into the materials library. I have tried inputting a marked up filament cost into the library, but it does not accurately account for machine time, Ie slicing a part at 0.1mm layers vs 0.28mm layers produces the same cost, however print time can be more than double. The cost of a machines hourly rate generally outweighs any cost of filament, so while the current cost calculator is great to easily see what a part costs in terms of filament usage, I think it offers little value.

### Affected users and/or printers

Companies/users who invoice printed parts to clients will benefit from this, also larger companies that may invoice parts internally, or even record costings for accounting purposes like machine depreciation over time used ect.

nallath, 2021-07-01:
I think this might be a duplicate of https://github.com/Ultimaker/Cura/issues/9261

fvrmr, 2021-07-01:
Hi @3Duke3 thank you for your feature request.
I would also see this as a duplicate of #9261 so I will close this issue as a duplicate.
Feel free to continue the thread there. Hope you understand!

closed 2021-07-01 by fvrmr