Ultimaker/Cura #10059: Play back speed feature

shonut260, 2021-07-01:
### Is your feature request related to a problem?

I can't speed up the play back.

No smoothness when rotating the part. It just jumps from position to position. Prusa is smooth, Fusion 360 and Mastercam are smooth. 

### Describe the solution you'd like

Give a speed of play back adjustment

### Describe alternatives you've considered

Using other software.

### Affected users and/or printers

Everyone will love having smoother transitions when rotating a part and most will like being able to speed up play back when verifying a print.

fvrmr, 2021-07-01:
Hi @shonut260 thank you for your feature request.
I will bring this up for discussion with the team, keep you posted!

Ghostkeeper, 2021-07-08:
We've decided to defer this issue. It would indeed be nice to have, but the performance we can muster here is limiting.

There are two main problems we'd need to tackle before we could implement this feature:

* The nozzle and layer view are scene nodes. Moving or changing them triggers a scene changed signal, which has quite a lot of listeners. Most of those will drop out quickly (e.g. the slicing backend will see that the nodes that changed are not nodes that will be printed, so it will not trigger a new slice) but it's still quite a lot of Python functions being called here. This would need refactoring or we wouldn't achieve a reasonable frame rate.
* We'd have to adjust the geometry shaders to be able to draw half lines. This isn't necessarily a performance problem but it could be a bit complex.

Because of this difficulty and the limited usefulness, the ratio of usefulness to work isn't very good. We're deferring this to communicate that you can't expect this to be implemented any time soon, unless priorities change.

closed 2023-07-13 by github-actions[bot]