Ultimaker/Cura #10061: Save File Name Wrong

TSBrownie, 2021-07-01:
### Application Version

4.10

### Platform

windows 10

### Printer

S3 Pro

### Reproduction steps

When I save the project under a different name, Cura does not remember that and next time I save it reverts to the original project name.  

### Actual results

File name does not change

### Expected results

If I save a project as Project01, then save a new version as Project02, the next time I go to save the current work, it should remember the most recent file name.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None needed.

fvrmr, 2021-07-01:
Hi @TSBrownie thank you for your report.
This is not really a bug but expected behaviour. However I do agree that this is confusing. So I will change this in a feature request and bring it up for discussion with the team. Keep you posted.

fvrmr, 2021-07-02:
I have discussed this with the team and created a ticket on our backlog for this. Devs see CURA-8358