Ultimaker/Cura #10062: Cura 4.10.0, Ultimaker S5, Error: Old version of Cura

discip, 2021-07-01:
**Application version**
Latest release 4.10.0

**Platform**
OS:
Win10 Enterprise [1909]

GPU0:
Intel HD Graphics 530

GPU1:
Nvidia Quadro K2200

**Printer**
Ultimaker S5

**Reproduction steps**
1. sliced object
2. saved to thumbdrive
3. tried to start the print

**Actual results**
Before starting the print, a messsge shows up stating:
```
The file is made with an older version of Ultimaker Cura. For optimal performance prepare you print job with Ultimaker Cura 4.4 or higher.
```

**Expected results**
  The above mentioned message should not appear.

nallath, 2021-07-01:
What version of the firmware are you using?

Ghostkeeper, 2021-07-01:
This is fixed in firmware version 6.4.0.

discip, 2021-07-01:
Ok, I see.
I am on 6.3.0.
Thanks.

closed 2021-07-01 by discip