Ultimaker/Cura #10063: The update to 4.10 do not slice whatsoever.

gaciap, 2021-07-01:
### Application Version

4.10

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

- After clicking the "Slice" button, there were no actions. Progress bar had not moved, and according to 
- If "Slice Automatically" was on, the progress bar was instantly jumping to ~15%. There were no actions after that.

### Actual results

The program did not act whatsoever. According to the task manager, it did not consume any resources.

### Expected results

It should slice.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I had to go back to the 4.9.1 version. The issue disappeared.

nallath, 2021-07-01:
Could you please provide us with the log files?

gaciap, 2021-07-01:
How to get those?

Regards:

*Przemyslaw D. GACIA*

Senior R&D Engineer, Sevendof <http://sevendof.com>

+47-469-33-221

<http://www.linkedin.com/in/przemyslawgacia>

fvrmr, 2021-07-01:
You can find your log files here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

gaciap, 2021-07-01:
I already uninstalled this version.
All the files in the directory were deleted.
I do not have logs anymore.

Regards:

*Przemyslaw D. GACIA*

Senior R&D Engineer, Sevendof <http://sevendof.com>

+47-469-33-221

<http://www.linkedin.com/in/przemyslawgacia>

fvrmr, 2021-07-02:
@gaciap then there is unfortunately nothing we can do about it. Since we can't reproduce this issue and can't see what was causing this issue.

ZoukiVinet, 2021-07-04:
I have this problem too. Maybe you can do something with my log file.
[cura.log](https://github.com/Ultimaker/Cura/files/6759795/cura.log)

fvrmr, 2021-07-05:
Hi @ZoukiVinet it seems like your antivirus is blocking Cura. Could you try slicing, with your antivirus turned off.

Ghostkeeper, 2021-07-08:
Well, the log file shows the following error:

    UM.Backend.Backend._runEngineProcess [147]: Couldn't start back-end: Operating system is blocking it (antivirus?): [WinError 87] De parameter is onjuist

The operating system is not allowing the slicing back-end to start. But what it says after is "invalid parameter", meaning that the operating system thinks the command to start the back-end is wrong, not really that an antivirus is blocking it.

This could occur if your preferences file contains an entry like:

```
[backend]
location =
```

That basically tells Cura that your slicing back-end is somewhere else than the default location. But if that entry is empty it would cause this sort of error. So I'm curious to see your `cura.cfg` file, now? Did you ever modify that?

ZoukiVinet, 2021-07-08:
I never modified anything. But somehow I fixed the issue by reinstalling cura.

Ghostkeeper, 2021-07-12:
Reinstalling Cura will override any changes made to Cura's Python source code. It would not change any configuration (unless you used the uninstaller to erase all configuration). So if that fixed it, perhaps the issue was a hard drive failure or a modified Cura installation.

Is the issue the same for @gaciap ?

closed 2021-07-23 by no-response[bot]