Ultimaker/Cura #10064: supports in new version - only TREE type?

jabbermacy, 2021-07-01:
### Application Version

4.10

### Platform

win 10

### Printer

mega S

### Reproduction steps

generate ANY supports gives only TREES, I don't want them

### Actual results

generate ANY supports gives only TREES, I don't want them

### Expected results

normal supports

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

generate ANY supports gives only TREES, I don't want them

Asterchades, 2021-07-02:
Change the Support Type, then. They would ordinarily default to "Normal" but you've got them set to "Tree".

jabbermacy, 2021-07-02:
YEP! My bad, I forgot to enable ALL settings on reinstall, thanks!

closed 2021-07-02 by jabbermacy