Ultimaker/Cura #10065: Raft (& Brim) doesn't work for bigger objects

JPN3D, 2021-07-01:
### Application Version

4.10.10

### Platform

Win10

### Printer

Own 330x330mm

### Reproduction steps

Just place bigger object with the hole(s) on the middle. 
Raft (or Brim, not 'only outside'). Slice

### Actual results

Raft & Brim create adhesion just on the edge of the object side & under  but not covering the whole object's area. 
If 'only outside' is ticked then Brim works as expected

### Expected results

Raft must cover the whole object's area + extra margin
Brim should be generated to cover holes are well

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_Kori315.zip](https://github.com/Ultimaker/Cura/files/6749010/CFFFP_Kori315.zip)

GregValiant, 2021-07-01:
This is a test isn't it...
The brim and raft don't see the large flat with the square holes in it because they aren't on the bottom layer.  There is a .25mm thick ring around the periphery.  Cura knew it was there and tried to account for it.  Since it is such a small feature even if you had turned on support (which it would need) it wouldn't have shown up unless you moved the part up off the build plate by 1 or 2mm.
This is the first layer on a raft.  It's as you described with no brim in the holes and the ring around the periphery is apparent.

[image]

This is the part buried .25mm into the build plate.  The brim is as you likely expected it should be.

[image]

This is not an easy part to print in one piece.  You could move the part down into the build plate .25mm and use a brim.  You could print the .25 thick "ring" separately and glue it on.  You can raise the part up off the build plate and use support.  I don't think a raft is going to work at all.

JPN3D, 2021-07-02:
Really sorry, my mistake. I had printed the part previously but now made a small change. I know why it had that outer bottom ring.

closed 2021-07-02 by JPN3D