Ultimaker/Cura #10066: Not retracting before every travel

ghoba82, 2021-07-01:
### Application Version

4.10

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

1. Sliced a sringing test
2. Ran it on printer

### Actual results

Not Retracting properly. Its hardly retracting after 4-5 layers and travels

### Expected results

it should retract before every layer and every travel

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info

fvrmr, 2021-07-02:
Hi @ghoba82 thank you for your report.
Could you share a project file (file - save project) and gcode?

3Duke3, 2021-07-03:
Have you checked "Maximum Retraction Count" and "Minimum Extrusion Distance Window", these settings are there to limit the amount of retractions to prevent filament grinding in the same spot. Retraction test towers require a very minimal amount of extrusion from side to side per layer, so its likely the above settings are preventing retractions taking place.

closed 2021-07-16 by no-response[bot]