Ultimaker/Cura #10067: Cura 4.10 force closes

robhp81, 2021-07-01:
### Application Version

4.10

### Platform

windows 10

### Printer

Creality 6 SE, Creality 6 Max (custom setting)

### Reproduction steps

Uninstalled new update, deleted all old files and settings,  restarted computer several times, reinstalled update 3 times.   I have update both video drivers and did a total system cleaning.

### Actual results

When selecting "open file" the program will force close by itself sometimes,  other times it will force close when you change the view from listing to "large icons",  or will force close when you select a file in the window when everything is in "listed view" and it tries to load it.

### Expected results

I should be able to select "open file" change the sub window view from "list view" to "large icon view"  and add file to the build plate for slicing.  

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[DxDiag 7.1.2021.txt](https://github.com/Ultimaker/Cura/files/6750991/DxDiag.7.1.2021.txt)

fvrmr, 2021-07-02:
Hi @robhp81 thank you for your bug report.
Do you maybe have Chitubox installed?

robhp81, 2021-07-02:
yes I do.

fvrmr, 2021-07-02:
Then it is a duplicate of https://github.com/Ultimaker/Cura/issues/9668
The problem is with Chitubox and you can uninstall the `CHITUBOX_Thumbnail.dll` file in the installation directory by following these steps:

- Open cmd with administrator rights
- Enter the command: regsvr32 / u "path (CHITUBOX_Thumbnail.dll path)"
- Restart the resource manager

I will close this issue as a duplicate but if you have more questions please add them to the other thread so we keep everything in one place.

closed 2021-07-02 by fvrmr

robhp81, 2021-07-02:
this doesn't work or solve my issue

robhp81, 2021-07-02:
[image]

Ghostkeeper, 2021-07-08:
That needs to be without the space between the `/` and the `u`. I think that was a typo.

`regsvr32 /u "<path to CHITUBOX_Thumbnail.dll>"`