Ultimaker/Cura #10068: Cannot restore backup / cannot save current backup

BirefringenceEffects, 2021-07-02:
### Application Version

4.9.1 & 4.10

### Platform

Windows 10

### Printer

Prusa i3 Mk3/Mk3s

### Reproduction steps

Hi guys, 
I'm in a bit of a pickle here.

I tried to uninstall the "Calibration shapes" Plugin from Cura via the GUI's marketplace (I was running 4.10 at the time), then I tried to reinstall it and it just keeps saying that the plugin fails to download. 
This has extended to all plugins as it would seem. 
I also cannot restore from existing backups or save a current backup. 
It almost seems like all connectivity to the Ultimaker server is restricted somehow....
 
After this I tried to roll back to version 4.9.1 as everything was working for me in that release. which installed fine, but the issues are consistent.
Not too sure what the problem is but it's pretty much broken Cura for me....

Kind regards, 
Jack.

### Actual results

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

I should communicate with the cloud properly and let me download plugins as well as upload backups etc...

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

logfile
[cura.log](https://github.com/Ultimaker/Cura/files/6752894/cura.log)

fvrmr, 2021-07-02:
Hi @BirefringenceEffects it looks like there is something wrong with the connection. Do you maybe have a firewall? I couldn't reproduce your issue.

BirefringenceEffects, 2021-07-02:
Hi @fvrmr, 

Nothing out of the ordinary just windows firewall I think

BirefringenceEffects, 2021-07-02:
I've just had a look at windows defender and it has been allowed through the firewall. 
So it shouldn't be an issue...
[image]

BirefringenceEffects, 2021-07-02:
current log file
[cura.log](https://github.com/Ultimaker/Cura/files/6753991/cura.log)

BirefringenceEffects, 2021-07-02:
current appdata folder zipped up with .cfg file removed for privacy.
[Uploading appdata_CURA.zip…]()

BirefringenceEffects, 2021-07-02:
[appdata_CURA.zip](https://github.com/Ultimaker/Cura/files/6754032/appdata_CURA.zip)

BirefringenceEffects, 2021-07-03:
Okie dokie, 

I've resolved this bug/issue; it turned out to be an out of date .NET framework for windows. 
After I updated to ".NET Framework 3.5 and 4.8 for Windows 10 Version 20H2 for x64", Cura was able to communicate with the server again and I've since been able to download plugins and backup profiles as well.

I'm making this bug as closed.

Thanks, 
Jack.

closed 2021-07-03 by BirefringenceEffects

fvrmr, 2021-07-05:
Hi @BirefringenceEffects thanks for the update!