Ultimaker/Cura #10069: Preview page job percentage step

ugurbu, 2021-07-02:
### Is your feature request related to a problem?

In fact it is not a problem. I wonder to see percentage of the job while looking at preview screen.

For example after slicing assume that 240 step is given. When I choose 120th step, there might be another information like %25 just near or some where 120 value.

So while tracking on 3rd printer screen, we can easily compare where the slicing process is on Cura.

[image]

### Describe the solution you'd like

I believe it is easy to integrate.

### Describe alternatives you've considered

no alternative needed.

### Affected users and/or printers

It will be helpful to compare with 3rd printer screen.

fvrmr, 2021-07-02:
Hi @ugurbu thank you for your feature request.
I will discuss this with the team. Keep you posted.

fvrmr, 2021-07-05:
Hi @ugurbu I have discussed this issue with the team and we decided to defer this issue. The reason why is we have got in the past related feature request to add height in mm etc there as well. Those were deferred because there is not much room to add more information. For the percentage it would also be quite confusing to point out if it is 50% of the print related to 3D print screen or 50% of the layers (120 layers of 240 layers).

closed 2022-05-01 by ugurbu