Ultimaker/Cura #10070: FLYING BEAR GHOST 5 WiFi

fortan25, 2021-07-02:
### Application Version

4.10.0

### Platform

Windows 10 64

### Printer

FLYING BEAR GHOST 5

### Reproduction steps

Установил плагин MKS WiFi. Плагин установился.

### Actual results

В настройках принтера не отображается кнопка настройки WiFi. Пробовал устанавливать-удалять. Ничего не помогает. В предыдущей версии всё работало без проблем.

### Expected results

Хотелось бы нормальной работы по WiFi

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-07-02:
Hi @fortan25 thank you for your report.
Could you share this in English. I assume that your wifi plugin is not working.

fortan25, 2021-07-02:
Good afternoon.1. Wi-fi is working on the printer-it sees the router and gets the address.2. What does the printer have to do with it, if after installing the plugin,   the Wi-fi settings button does not appear in the printer settings ?Everything worked fine on the previous version of Cura.

Sladerix, 2021-07-06:
I've problems too.
Cura is at version 4.10.0, MKS is at version 1.2.2 but in the Printer Preferences i don't see anymore the button "MKS Wifi"
[image]

fvrmr, 2021-07-06:
It is not an issue with Cura but with the plugin. The plugin is only supported till 4.9 See: https://github.com/Jeredian/mks-wifi-plugin/issues/171

I will close this issue since this is not something we will fix.

closed 2021-07-06 by fvrmr

SupersADF, 2021-08-14:
if the plugin doesn't work follow these steps:
1) open plugin.json in folder C:\Users\$USERNAME\AppData\Roaming\cura\4.10\plugins\MKSWifiPlugin\MKSWifiPlugin, where $USERNAME - you user name
2) edit "maximum_cura_version": "4.9" to "maximum_cura_version": "4.10"
3) add "7.6.0" in "supported_sdk_versions" like this
 "supported_sdk_versions": [
        "7.1.0",
        "7.2.0",
        "7.3.0",
        "7.4.0",
        "7.5.0",
        "7.6.0"
    ],
4) run Cura and config your printer!