Ultimaker/Cura #10072: Inconsistent quality profile names

santanig5, 2021-07-02:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

Creality CR-10S Pro

### Reproduction steps

A designer Shark Aero provided Cura print profiles based on their printer, the profile name is "normal" when I attempt to import into Cura 4.10.0 for my Creality printer I get an error as "normal" is not a profile listed for Creality.
If I add a Ultimaker printer I am able to import the profile successfully, but still cannot see it for Creality.
Is there a way to address this issue or a work around?

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/6755251/cura.log)

fvrmr, 2021-07-05:
Hi @santanig5 this has to do that Creality profiles don't have a normal quality type and Ultimaker printers do. 

In the curaprofile file you want to import, quality type is set to normal. And in this case Creality doesn't have a normal profile only super, dynamic, standard and low quality. 
Do you know for what printer these profiles were made?

Here you can read more about importing profiles: https://github.com/Ultimaker/Cura/issues/8229

santanig5, 2021-07-05:
Hi,
I do not know what type of printer these profiles were made from, would the gcode have that info?
Knowing that these came from a full scale airplane designer I assume a commercial grade printer like Ulitmaker.
Why are Creality profiles non-standard?
Who chose the profile names?
Can’t they be merged in a future version to eliminate this issue?
I followed the link for case #8229, however that appears to be a material issue, I don’t see how it pertains to this issue.
Kind regards

Ghostkeeper, 2021-07-09:
So to summarise: Each quality profile has a "quality level". Profiles with the same quality level are expected to produce a similar result, even though they might have different settings because they specialise for a certain material/nozzle/printer combination. Custom profiles are changes on top of these quality profiles; if you load the custom profile, it'll find the quality level that the custom profile was created with and apply the changes that the user stored on top of that.

The custom profiles you can select from are only the ones for which the correct quality level is available. For instance, if you created a profile while you had a 0.25mm nozzle active, based on the "super-extra detailed" quality level, this profile won't be available if you switch to a 1.0mm nozzle, because that nozzle can't print super-extra detailed.

When you export a custom profile, it only stores the changes that you made in the actual profile, and the quality level that the profile is based on. When you import it again, it checks if the quality level is available in the current configuration. If the quality level is not available for the printer at all, you get this message.

> Why are Creality profiles non-standard?
> Who chose the profile names?

These quality levels have an identifier. Ultimaker has chosen `normal`, `fast`, `fine`, `draft` and such for its own profiles and also for the global profiles. However the Creality profiles use `low`, `standard`, `adaptive`, etc. for theirs. This was chosen by the [people who designed those profiles](https://github.com/Ultimaker/Cura/pull/5809/files). They had slightly different requirements I think, with the adaptive profile, but it could be that it was also explicitly designed such that profiles from different printers couldn't accidentally be imported for Creality printers, to which they would be incompatible.

> Can’t they be merged in a future version to eliminate this issue?

To merge them in a future version we'd have to map all existing profiles users may have in their configuration folders to a new quality level, and rename the quality levels bundled with Cura. This would increase compatibility if a user would want to move a custom profile from one printer to another. Since the profiles on the hard drive of every Creality user would have to be changed, this requires a version upgrade. Ultimaker doesn't generally adjust profiles for third-party printers. We test if they load in properly, but otherwise just provide them as-is to the user. So if this is desired, the Creality community would have to implement a version upgrade to map one quality level to another. I can't test which quality level would be similar to which since I don't have your printer model.

> I do not know what type of printer these profiles were made from, would the gcode have that info?

G-code files generated by Cura have a slightly-readable listing at the bottom which profiles were used and which settings were overridden. It's not enough to reconstruct the entire configuration but you can see which printer profile was used, at least if the profiles used were specific to a printer.

santanig5, 2021-07-09:
Thanks for the explanation, although most of it is over my head it seems like allot of work would need to be done.
Would it be possible to create a utility/tool to do the conversion?
I have seen some procedures that involve renaming and zipping the files to change the quality name which I have done successfully but is very tedious.
BTW, I have a Creality CR-10S Pro.
Kind regards

Sent from my Verizon, Samsung Galaxy smartphone

Ghostkeeper, 2021-07-13:
> Would it be possible to create a utility/tool to do the conversion?

Yes, that's possible.
But effectively, this is what the version upgrade plug-ins are, too.

GregValiant, 2024-11-19:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

santanig5, 2024-11-19:
I haven't come across the issue in quite some time, and I've learned workarounds since this was open. 

Thanks!