Ultimaker/Cura #10073: Cura 4.10 force closes

robhp81, 2021-07-02:
### Application Version

4.10

### Platform

windows 10

### Printer

Creality 6 SE, Creality 6 Max (custom setting)

### Reproduction steps

Uninstalled new update, deleted all old files and settings, restarted computer several times, reinstalled update 3 times. I have update both video drivers and did a total system cleaning.  I tired what member fvrmr said to do and keep getting errors!!!  This person then closed out the ticket and said there is another with the same problem and told to write there!!  

### Actual results

When selecting "open file" the program will force close by itself sometimes, other times it will force close when you change the view from listing to "large icons", or will force close when you select a file in the window when everything is in "listed view" and it tries to load it.
[image]
[image]

### Expected results

I should be able to select "open file" change the sub window view from "list view" to "large icon view" and add file to the build plate for slicing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

keep getting error message

CQCDoug, 2021-07-02:
Having the same problem with both 4.9.1 and 4.10.0......

3Duke3, 2021-07-03:
Do you guys also have Chitubox installed? Have you seen #9668

fvrmr, 2021-07-05:
Hi @robhp81 I closed the issue because it is a duplicate of an other issue. We do this to keep everything on one place.
Your issue has to do with Chitubox, that is causing this problem. The only work around we have for this is what I suggested before was a solution from the Chitubox forum: https://forum.chitubox.com/?thread-419.htm
Since this is not a problem of Cura but a problem causing from Chitubox. 

Looking at your screenshot you need to use /u there is a space between them and the location of the dll is not correct. You need to find the location in your program files where Chitubox is listed. 

Let me know if you need more help!

closed 2021-07-19 by no-response[bot]