Ultimaker/Cura #10075: (4.10.0) Platform is lowered slightly after prime blob, causing print to not stick

mothnox, 2021-07-03:
### Application Version

4.10.0

### Platform

Manjaro Linux

### Printer

Creality Ender 5 Plus

### Reproduction steps

1. Load and slice model
2. Print model

### Actual results

After printing a prime blob (on by default and cannot be turned off, see #10074 ), the platform is lowered slightly before the printer returns to the center to begin printing, resulting in the print not sticking. 

### Expected results

The platform would not be lowered, or would raise back to the proper height before beginning to print the model. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

This isn't due to the printer not being configured properly, as I was previously using Creality Slicer and had no issue with print adhesion. The prime blob itself also sticks just fine. I'm not certain whether or not this would occur if the prime blob was not printed, as I'm currently unable to disable it. 
[CE5P_supports_test.3mf.zip](https://github.com/Ultimaker/Cura/files/6757679/CE5P_supports_test.3mf.zip)

mothnox, 2021-07-03:
I was able to get it to print correctly by removing the line of gcode that changed the Z height after printing the prime blob.

GregValiant, 2021-07-03:
From FDMPrinter.def.json (the basic printer definition file)...
"prime_blob_enable":
**"default_value": false,**
                    "settable_per_mesh": false,
                    "settable_per_extruder": true,
                    **"enabled": false,**
There is no mention nor overrides of Prime Blob in the CrealityBase.def.json file or in the Creality_Ender5Plus.def.json file.  It does get enabled for Dual Extruder printers but the default is still "false".

I sliced the 3mf file and your startup gcode does have the purge lines in it but there is no blob.  This is the gcode prior to the start of the first layer:
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
**G1 Z2.0 F3000 ;Move Z Axis up** (bed down on an Ender 5)

G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:126
;LAYER:0
M107
M204 S500
M205 X8 Y8
G0 F7200 X159.742 Y167.307 **Z0.2**
You can see in the last line that the Z = 0.2mm and that's what you have in the Cura settings as Initial Layer Height.  You may need to check your ABL settings.  Your Startup Gcode currently uses M420 S1 Z2 as the ABL command and possibly the Z fade height (or offset) is incorrect.
The "G1 Z2.0 F3000 ;Move Z Axis up" line is in your startup gcode and can be altered to **G1 X10 Y10 Z12.0 F3000**.  That will move the nozzle up and away from what you are referring to as a blob but is actually ooze.  It won't correct the distance from the bed for the first layer though.

mothnox, 2021-07-03:
Ah, I mistakenly thought that the purge line was a prime blob. I'd done some searching to try to find what that line was called so I could turn it off in the settings, and the only thing I could find about that was the "prime blob", so I thought it was that. That said, the bed was still being lowered (the be moves on this printer, not the nozzle) after the purge line, which resulted in the nozzle being slightly too far from the bed to stick once it started on the actual print.

GregValiant, 2021-07-03:
Go to Manage Printers and then Machine Settings.  Then put semi-colons in front of the purge line commands to turn them off.  If you change the G1 Z2.0 F3000 line to G1 X10 Y10 Z10 F3000 the nozzle won't get stuck in the ooze.

G28 ;Home
**;M420** S1 Z2 ;Enable ABL using saved Mesh and Fade Height

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
**;G1** X10.1 Y20 Z0.28 F5000.0 ;Move to start position
**;G1** X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
**;G1** X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
**;G1** X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
**G1 Z2.0 F3000 ;Move Z Axis up**

mothnox, 2021-07-03:
What do you mean about the nozzle getting stuck in the ooze? That's not something that's happened, as far as I'm aware.

GregValiant, 2021-07-03:
My start gcode is much the same as yours.  The nozzle would lift 2mm before it retracted, Then it would sit back down in the little puddle puddle of ooze and drag a long string across the area where the print would go.  If yours doesn't do that then you don't need to change that line.

mothnox, 2021-07-03:
No, mine doesn't do that. It moves off to the side, away from where the print will be, and extrudes a line a little ways along the Y axis and back. Then the platform lowers slightly, and it returns to the center to begin the print. But the platform doesn't raise back up after that, causing the nozzle to be slightly too far for the print to stick to the bed when it begins printing.

GregValiant, 2021-07-03:
Post a Gcode file here that shows the annoying movement.
When I slice your 3mf file it doesn't show any drop, just a movement to Z=0.2 and then it starts on the brim.  

I don't have an ABL but I seem to remember that if a mesh has not been defined within the printer for M420 then bad things can happen.  If you have M420 disabled then how are you leveling the bed?

mothnox, 2021-07-03:
I attached two files, the first is one generated by Cura, with no edits. The second one I edited, removing the "G1 Z2.0 F3000" at the end of the purge line instructions, which fixed the problem for me. Is it possible that the Ender 5 Plus (or maybe just certain versions of its firmware) interpret some instructions differently than expected? 

As for ABL, I get the feeling we might not be talking about the same thing? I was referring to a setting on my printer itself, for auto-leveling. I level the bed by measuring via BLTouch, making manual adjustments as needed, and then measuring again. It's worked very well for me so far, if a bit slower than it would be with auto-leveling. 
[screwhole_test.gcode.zip](https://github.com/Ultimaker/Cura/files/6759033/screwhole_test.gcode.zip)
[screwhole_test_fixed.gcode.zip](https://github.com/Ultimaker/Cura/files/6759034/screwhole_test_fixed.gcode.zip)

GregValiant, 2021-07-04:
The Creality firmware is no longer exactly Marlin.  They took the Marlin firmware, altered it, and didn't roll it back into Marlin.  That makes it a proprietary firmware.  The people at MarlinFW.org are not happy.
Within the firmware is an option for TFT style LCD's.  I think Creality missed it on at least the Ender 3 and 3Pro.  The result is that commands that send messages to the screen (the M0 pause command sends "click to resume") don't work in the Creality versions.  So it isn't that the firmware is interpreting things "differently" but rather "not at all".
Since your StartUp Gcode doesn't have any of those type of commands (M0, M1, M117) then the printer is just going where it is told to go.  The problem with that G1 Z2.0 line was that the build plate dropped BEFORE the retraction (that occurs 3 lines later).  Getting rid of that line allowed the retraction to happen when the nozzle was still at the end of the previous extrusion at Z=.28 instead of being in the air at Z=2.0 and oozing.
Regarding  ABL - have you tried using G29 instead of M420?  I don't know ABL systems but a lot of Creality machines use G29.

mothnox, 2021-07-04:
I did some reading, and it looks like it should be possible to build Marlin for the Ender 5 Plus. Might give it a try since I'm out of filament at the moment anyway. 
I haven't tried using G29, no. The day that I opened this issue was actually the first time I'd looked at or modified gcode, so I'm kind of figuring things out as I go. I'll look into that.

GregValiant, 2021-07-04:
Gcode is a simple machine movement language.
If you don't know what you're doing then firmware is a black hole.
The list of Gcode commands is here [https://marlinfw.org/meta/gcode/](url)

mothnox, 2021-07-04:
Yes, I was referencing that already. When I said I was figuring things out as I go, I meant I looked up what various commands did as I encountered them.

fvrmr, 2021-07-05:
> as I was previously using Creality Slicer and had no issue with print adhesion

Also good to know Creality Slicer is forked from Cura, so it could be working in a previous version of Cura which Creality Slice was forked from but not in the newer ones.

mothnox, 2021-07-05:
From the sound of things, it seems more likely that it's due to Creality Slicer making changes or adjustments to account for their printers' deviation from Marlin.

mothnox, 2021-07-06:
I've tried G29, but it doesn't seem to make any noticeable difference. It _has_ resulted in the printer extruding a line from the purge line to the center for some reason. The printer settings for auto-leveling also seem to override the gcode. G29 with the printer's auto-leveling setting off does nothing, and with auto-leveling on but no G29 specified, the printer probes in its default 4x4 matrix before each print. I'm not sure if the printers auto-leveling settings also override M420.

GregValiant, 2021-07-07:
Here is a little light reading for you [https://community.ultimaker.com/topic/31676-how-do-i-turn-off-this-initial-triangle/page/3/](url).
Maybe you'll find something that will help.

mothnox, 2021-07-07:
Yeah, I've seen solutions for the triangle thing before. I'm not really concerned about it at the moment, though, since it's not generally in the way. Mostly I just want to find a way to get my prints to stick properly with Cura. At this point I'm probably just going to slice the same model with Cura and Creality Slicer, and look at the gcode to see what's being done differently.

Ghostkeeper, 2021-07-09:
Perhaps it's worth a try to save a project file from Creality Slicer and then open that project file in Ultimaker Cura?

If you would like to propose changes to profiles, we can incorporate them. Preferably via a pull request, but just a text message works just as well.

mothnox, 2021-07-12:
Some stuff has come up recently that I've had to deal with, so I've been pretty busy. That said, I did figure some stuff out, so I'll submit a pull request once I have time.

MariMakes, 2023-05-24:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, 

I stumbled upon your issue with the primeblob not sticking, @mothnox, can you let us know if your issue is resolved?

Would you like just to take another look? 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-06-08 by github-actions[bot]